JdeRobot/base #669: [drivers] OpenNiServer does not work with libfreenect.

fqez, 2017-01-17:
This is an issue for @chanfr mainly. We have installed OpenNiServer in theta02 (in the Uni Laboratories) and we linked it with libfreenect (manually). The kinects work with NiViewer examples, but not with our drivers. I've managed to make it work sometimes deactivating the Depth or the RGB camera, but it does not work with the two of them simultaneously.
The trace is the following:

```
OpenniServer: Selected device: 
OpenNI2-FreenectDriver: Using libfreenect v0.5.3
OpenNI2-FreenectDriver: Found device freenect://0
OpenNI2-FreenectDriver: Opening device freenect://0
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
Violación de segmento (`core' generado)
```
This trace comes when you try to launch the server with both RGB and DEPTH cameras active at the same time in video mode 0 (320x240 30fps)

If you launch the same scenario with mode 4 (640x480 30fps) sometimes it "works" and sometimes it doesn't. When it works the trace is the following:

```
OpenniServer: Selected device: 
OpenNI2-FreenectDriver: Using libfreenect v0.5.3
OpenNI2-FreenectDriver: Found device freenect://0
OpenNI2-FreenectDriver: Opening device freenect://0
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
[Stream 70] Invalid magic edff
[Stream 70] Invalid magic eeff
write_register: 0x0005 <= 0x01
[Stream 70] Invalid magic 570a
[Stream 70] Lost 7 packets
[Stream 70] Lost 7 total packets in 4 frames (1.750000 lppf)
[Stream 70] Lost too many packets, resyncing...
[Stream 70] Expected max 1748 data bytes, but got 1908. Dropping...
[Stream 70] Invalid magic ffff
[Stream 70] Invalid magic 97ba
[Stream 70] Invalid magic a9b5
[Stream 70] Invalid magic ffff
[Stream 70] Invalid magic 4fc9
[Stream 70] Invalid magic e6fc
[Stream 70] Invalid magic 59cb
[Stream 70] Invalid magic b5f8
[Stream 70] Invalid magic 6f6d
[Stream 70] Invalid magic ffff
[Stream 70] Lost 12 packets
[Stream 70] Lost 19 total packets in 8 frames (2.375000 lppf)
[Stream 70] Lost too many packets, resyncing...
[Stream 70] Invalid magic cada
[Stream 70] Invalid magic 9eb3
[Stream 70] Invalid magic 9c23
[Stream 70] Invalid magic d9fb
[Stream 70] Invalid magic 06e0
[Stream 70] Invalid magic ffff
[Stream 70] Invalid magic beb8
[Stream 70] Invalid magic 686d
[Stream 70] Invalid magic 9716
[Stream 70] Invalid magic fc7f
write_register: 0x0047 <= 0x00
write_register: 0x0047 <= 0x00
[ERROR] OpenniServer: error at set depth videoMode: 	Stream setProperty(3) failed

[INFO] OpenniServer: depth modes
fps: 30x: 640y 480
[ERROR] OpenniServer: error at set color videoMode: 	Stream setProperty(3) failed

write_register: 0x0005 <= 0x00
[Stream 70] Invalid magic ffff
[Stream 70] Lost 1 packets
[Stream 70] Lost 20 total packets in 11 frames (1.818182 lppf)
[Stream 70] Expected max 1748 data bytes, but got 1908. Dropping...
[Stream 70] Invalid magic ffff
[Stream 70] Lost 5 packets
[Stream 70] Lost 25 total packets in 11 frames (2.272727 lppf)
[INFO] OpenniServer color modes:
fps: 30x: 640y 480
write_register: 0x0006 <= 0x00
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
COLOR: fps:0w640h480
DEPTH: fps:30w640h480
[ERROR] OpenniServer: error at set syncronization: 
[INFO] Starting device pre-heating .... 
End device pre-heating
Creating camera cameraA
Starting thread for camera: cameraA
              -------- openniServer: Component: CameraRGB created successfully(default -h 0.0.0.0 -p 9999:ws -h 0.0.0.0 -p 11000@cameraA
Creating camera cameraB
Starting thread for camera: cameraB
              -------- openniServer: Component: CameraDEPTH created successfully(default -h 0.0.0.0 -p 9999:ws -h 0.0.0.0 -p 11000@cameraB
OpenniServer initialized
```

An so on... can you check it please? This is useful for the 3D People Tracker!

Thanks!

chanfr, 2017-01-22:
Confirmed.

kinect works for me on:
- freenect viewers
- openni2  viewers

But not with openniServer

chanfr, 2017-01-29:
Can you test if this is now working? 
I made a big refactor of the component in my branch frivas_devel. I tested both with kinect and xtion and it seems to work.

fqez, 2017-01-29:
Yes, of course!!

I will test it tomorrow. Thanks a lot!

fqez, 2017-01-30:
Confirmed, it works now in every execution. Video modes is no longer supported, but it works so it's super fine by me :)
We still need to tell openniServer where the libfreenect libraries are, but is not a big concern at the moment.
Here's a video showing the refactorization!

https://youtu.be/kJh-7ul6ilY

closed 2017-02-06 by fqez