JdeRobot/base #677: All our components should use SI

aitormf, 2017-01-31:
All our components should use SI (International System of Units).

This is going to help to know that units are the interfaces, because some, such as the laser, use millimeters, not are exactly in that units are shipped speeds the kobuki and drones.

Also going to give less problems of compatibility with ROS, because they already used the SI

Drivers:
ardrone_server
Plugins:
car
f1
flyingKinect2
kinect
quadrotor2
turtlebot

shivangg, 2017-03-23:
I'd like to work on this issue. How should I start the conversion to SI units? Should the conversions to SI units be explicitly coded or something like (boost units) may be used?

shivangg, 2017-03-24:
I have trouble finding where the basic_component is using any units(which shall be convert edto SI). It only uses the camera ICE interface. If  @aitormf can point me to the right direction.

jmplaza, 2017-03-26:
Hi @shivangg AFAIK there is no explicit description of the units in the code. If you don't have real robots you can check the units using the simulator. Typically the JdeRobot applications (like tools) and the drivers exchange numbers, like movement orders, through ICE interfaces (or ROS-messages). Those numbers have to be checked to mean SI units. For instance, when using the Formula1 car: 3 m/s for V orders or 1 rad/s for W orders. You can test a world in Gazebo and check there than the Formula1 advances at 3 m/s when a 3 is sent from KobukiViewer, or check that a drone rotates 1rad/s when ordered 1 from uavviewer....

Cheers,

reysam93, 2017-06-11:
VisualHFSM shouldn't be in the list.. The tool doesn't use any units by itself. The units used in it depends only on the drivers that will be used.

jmplaza, 2017-06-12:
You are right. I've just deleted the item for VisualHFSM in that list

closed 2021-02-21 by jmplaza