JdeRobot/base #690: [drivers] New Formula1 race circuit, similar to Montecarlo one

jmplaza, 2017-02-21:
It will be used in several exercises of [Teaching Robotics](http://jderobot.org/Teaching_robotics_with_JdeRobot#Exercises_on_autonomous_cars)

@kb8alvaro has built good prototypes for this, check them at his [webpage](http://jderobot.org/Avillamil-tfg)

[image]

kb8alvaro, 2017-03-08:
It is done!! Just waiting to know how to do the pull request and I'll upload so you can race :)
[image]

fqez, 2017-03-08:
Hi @kb8alvaro !

you have the instructions to make your own pull requests here: https://github.com/JdeRobot/JdeRobot/wiki/How-To-Contribute

jmplaza, 2017-03-30:
Hi, I've tested the world in JdeRobot-5.5 and the Monaco circuit just falls down as shown in this [video](https://youtu.be/z7bj_QdEWMQ)

aitormf, 2017-04-18:
merged #724

closed 2017-04-18 by aitormf