JdeRobot/base #698: gazebo naoworld.world does not work as it should

saifullah3396, 2017-03-02:
There is no nao model available in the current source code for this world (naoworld.world). It gives the following error:
Error [parser.cc:581] Unable to find uri[model://nao]

jmplaza, 2017-03-02:
Current release of JdeRobot does not support Nao humanoid robot. We focus on Kobuki robot, RGBD devices (like Kinect), drones (ArDrone from Parrot, Solo drone from 3DR), several cameras and simulated cars (formula1, taxi...). Previous releases, like JdeRobot-5.2, already provided support for [Nao humanoid](http://jderobot.org/Fperezl) in Gazebo-5 but it has not been ported to Gazebo-7 yet. 

It is not planned include support for Nao in current release but we are open to collaboration on this :-).

saifullah3396, 2017-03-04:
I will be happy to collaborate on adding the support for naoqi in the current release of the software. I will start the code review for the previous releases and will update you on any progress that I make.

jmplaza, 2017-03-06:
Hi @saifullah3396 that would be very nice!. It is not on the [roadmap](http://jderobot.org/Main_Page#Roadmap) of JdeRobot right now, and so, out of our GSoC-2017 participation, but we would help in extending the Gazebo plugins to include again support for Nao humanoid. You can find interesting links for this [here](http://wiki.ros.org/nao)

saifullah3396, 2017-03-08:
Hey @jmplaza, thank you for the response. I have been a Nao developer myself and so I am particularly interested in the development of an easy-to-use simulator or interface for Nao other than ROS. I will look at the provided links.

closed 2018-02-21 by fqez