JdeRobot/base #717: problems installing jderobot-deps-dev, opencv-dev .

MohmadAyman, 2017-03-11:
Hello,
I'm trying to install jderobot the development version and I've ros-kinetic installed from source
this is what happens when I try to install 
```
jderobot-deps-dev
``` 
```
Unpacking opencv-dev (3.1.0-1407-gfa87422) ...
dpkg: error processing archive /var/cache/apt/archives/opencv-dev_3.1.0-1407-gfa87422_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/opencv_createsamples', which is also in package libopencv-dev 2.4.9.1+dfsg-1.5ubuntu1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
```
do I have remove opencv2? that might cause problems to some ros pakcages.
thanks.
I'm actually trying the dev version because I'm looking forward to understand the project and apply for a GSoC project this summer.

aitormf, 2017-03-13:
Hi @MohmadAyman, 

I recommend you wait for the new version of jderobot that we are going to take this week. it comes with support for ROS and change some dependencies as opencv (#708, #709 )

regards, Aitor

Sayan98, 2017-03-14:
I was also facing some problems with installation especially with kobuki which has dependency sophus (make failing at 91%) , it was complaining of some missing cpp header files. Looking forward to the new release.

fqez, 2017-03-14:
Hi @Sayan98 

yes, we are working on the kobukilib library also. This week you will be able to compile it and install it without any kind of trouble. We will let you know as soon as we get it!

Cheers and sorry for the issue.

MohmadAyman, 2017-03-15:
Okay, thanks.

fqez, 2017-03-15:
Hi all

the new release is now out (JdeRobot 5.5). You can try to install it and report your problems here. To install this version we get rid of some of our 3rd party libraries: pcl, vtk and opencv. Now we depend only on ROS and its dependencies.
In order to install this version you have to:

1. Uninstall pcl, vtk and opencv libraries installed by JdeRobot-5.4.2 and the actual jderobot package with ```apt purge <package>``` and ```apt autoremove```
2. Update JdeRobot repositories.
3. Install the jderobot package as usual.

Cheers

MohmadAyman, 2017-03-17:
@Sayan98  I had the same problem with the  ```sophus/se3.hpp``` missing as you,
I worked around it by searching for ```se3.hpp``` manually and replaced its includes in the ecl_linear by the absolute path of the library,
e.g.  #include <sophus/se3.hpp> to #include "/home/hgfj/.../sophus/se3.hpp".

aitormf, 2017-03-17:
@MohmadAyman @Sayan98 
the error with sophus is solved by installing sophus package:
<pre>
sudo apt install sophus
</pre>
It is a dependency of kobukilib package but it didn't install sophus. Now kobukilib install sophus also

MohmadAyman, 2017-03-17:
that explains why i found sophus only in the ros libs, thanks.

MohmadAyman, 2017-03-20:
hello, i've successfully installed the library, and i could run the cameraserver and the cameraviewer, but i could only run them when i'm their directory, unlike the examples where you just run ```$ cameraviewer cameraviewer.cfg``` 
is the examples just assuming that i'm writing the path of the file or do I have something incomplete in my installation? e.g. i run it like ``` $  ./home/...../drivers/cameraserver/cameraserver <path_to_file>/cameraserver.cfg```

jmplaza, 2017-03-20:
Have your executed "sudo make install"? It seems that the executable files have been properly generated but not installed in the general paths of your machine. For instance, when installing JdeRobot from debian packages those files are copied at /usr/local/bin/

MohmadAyman, 2017-03-20:
Your right! -and sorry for the silly question- it seems I did forget to run the ```sudo make install``` , and thanks for all the help.

Sayan98, 2017-03-28:
@aitormf Sorry for my late reply! The latest release built successfully, and I've everything up and running. Although I'd like to let you know that the `CamerView` and `CamerServer` have a weird timeout warning, which is a pretty common issue from what I observe.

LujunWeng, 2017-03-28:
@Sayan98 I got the same time out warning issue. But, anyway, camera works. Is there a special open 'issue' for this problem?

Sayan98, 2017-03-28:
@LujunWeng I don't know about any special open issue. Also, the timeout issue is quite common, many of the examples in the website also have the same issue. And yeah, confirmed, that the camera works fine.

Santi-7, 2017-03-28:
@Sayan98 Maybe you are working at a higher frame ratio than the one supported by your camera for the specified resolution?

Sayan98, 2017-04-01:
To anyone else visiting this thread, the latest release works fine. Alternately, you could use the docker image of JdeRobot, to use it just type in `docker pull jderobot/jderobot`.

closed 2017-04-01 by jmplaza