JdeRobot/base #727: [compile] Error while compiling JdeRobot

Daniel1108, 2017-03-14:
Hi,
I'm trying to install JdeRobot from **https://github.com/RoboticsURJC/JdeRobot**
But i'm having problems with compilation. All components are compiled but cameraviewer get me this error:
/home/user/JdeRobot/src/tools/cameraview/cameraview.cpp:28:42: fatal error: jderobot/comm/cameraClient.hpp: No existe el archivo o el directorio **(file not found)**

Thanks

yayaneath, 2017-03-14:
Hi there!

I'm also beginning with JdeRobot but I think I can help with this one. As far as I'm concerned, this error may be being thrown because you are building the _cameraview_ tool without having ROS installed in your computer. Due to this fact, _jderobotcomm_cpp_ is not built and then _cameraview_ fails.

fqez, 2017-03-14:
Hi!
yes @yayaneath is right, you cannot compile cameraview nor kobukiViewer because you must have ROS Kinetic installed in your system before. We are preparing the repo for the ROS compatibility yet. This week (most likely tomorrow ) we will release the new version of the packages (JdeRobot v5.5) and the new installation instructions to make the project compile easily.
Sorry about your issues, we are working hard in this version. We will let you know when it is ready as soon as we release it.

fqez, 2017-03-15:
Hi all

the new release is now out (JdeRobot 5.5). You can try to install it and report your problems here. To install this version we get rid of some of our 3rd party libraries: pcl, vtk and opencv. Now we depend only on ROS and its dependencies.
In order to install this version you have to:

1. Uninstall pcl, vtk and opencv libraries installed by JdeRobot-5.4.2 and the actual jderobot package with ```apt purge <package>``` and ```apt autoremove```
2. Update JdeRobot repositories.
3. Install the jderobot package as usual.

Cheers

GYengera, 2017-03-16:
I am working on Ubuntu Trusty (14.04 LTS). Will JdeRobot work with ROS Indigo or is ROS Kinetic necessary?

jmplaza, 2017-03-16:
ROS Kinetic is necessary. 

ROS compatibility has been hard to achieve. JdeRobot works fine with ROS Kinetic and should work with previous ROS releases too, like Indigo or Jade, but we have not tested. You can install Ubuntu Xenial (16.04 LTS) on a virtual machine and there the brand new JdeRobot 5.5 debian package, which automatically installs official ROS Kinetic packages from OSRF.

shivangg, 2017-03-16:
@GYengera Ubuntu Trusty had many issues(even though I had ROS indigo installed), but the installation went smoothly on Ubuntu Xenial(with ROS Kinetic)

GYengera, 2017-03-18:
I upgraded my system to Ubuntu Xenial. `cmake` has been executed successfully, but `make` command fails. I think the issue is caused due to jderobot-deps-dev having unmet dependencies: zeroc-ice-utils and opencv-python. When I try to install these packages certain errors occur. Has anyone else experienced this issue and know how to fix it? If not then I think I'll go ahead and install JdeRobot on a virtual machine.

shivangg, 2017-03-18:
This problem might be same as [this](http://jderobot.org/Manual-5#Installation_for_developing_in_JdeRobot). Also, for me, the fresh install of Xenial(not upgrade from Trusty) along with ROS kinetic is what worked smoothly. Tested successfully with JdeRobot 5.4.2 and 5.5 .

aishik-pyne, 2017-03-18:
I had made a mistake by installing a non LTS version which caused ROS installation problems. Fresh installation of Xenial would do the trick.

LujunWeng, 2017-03-19:
I experienced the same issue as well as the issue with "opencv headers not found". I solved these issues by run the following commands after installing dependent packages and before compiling jderobot:
```
echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc 
source ~/.bashrc 
```
It works well. But it comes from the section of "Installation for developing in JdeRobot" in Manual-5. It should be in the section of "Installation for developing in JdeRobot", just before the instruction of compiling the project.

jmplaza, 2017-03-19:
I've just updated the web page with the doc, just adding those lines to the "Installation for developing in JdeRobot" section.

jmplaza, 2017-04-01:
Solved with new JdeRobot release 5.5 and some post-install instructions, which have been added to the web doc. JdeRobot now compiles fine in the reference distribution Ubuntu 16.04.

closed 2017-04-01 by jmplaza