JdeRobot/base #729: Make roscpp required

LujunWeng, 2017-03-20:
Hi,

I am just wondering why ros is not a required package. 

In `JdeRobot/Deps/ros/CMakeLists.txt`, 'QUIET' is used:
```
find_package(roscpp QUIET)
```

Would it be better to make it REQUIRED so that if ros is not configured properly, `cmake` will fail and `make` will not be executed before ros is configured and cmake passes?

aishik-pyne, 2017-03-22:
Hey Lujun,

I had the same problem a few days ago and `cmake` passed with ROS directories NOT FOUND.
But after that during `make` it failed as the linkers didn't bind and a C++ package was not found `FATAL ERROR` was shown.

So even i think it's a good idea to make the ROS packages `REQUIRED`

fqez, 2017-03-22:
Yes, you both are right. This is set as ```QUIET``` because till the last week, we were on a testing stage of the ROS compatibility. We will change it to ```REQUIRED``` as soon as we can. Thanks for the report.

chanfr, 2017-03-25:
Why don't we leave ROS as optional and just disabling ROS compatibility when ROS is not available?
This makes jderobot more flexible when using it on embedded systems for instance. 
See the PR.

lr-morales, 2017-04-04:
I agree with the optionality. JdeRobot is (or is taking steps to be) ROS compatible, not ROS-required. 
As long as the core part of JdeRobot not depends on ROS-related components, I think that this is a good approach. Also that would mean marking and disabling/modifying new ROS-dependent components to check the availability (as intended with PR #737).

I would add an option to allow disabling ROS manually, similar to the one that allows to disable components. What do you think?

aishik-pyne, 2017-04-04:
In my opinion users who are compiling for the first time ROS option is skipped and C++ linkers are not linked and while executing make it gives new users a hard time.

So if ROS dependency is made optional clear instructions should also be given as how to install only the non-ROS dependent parts.

chanfr, 2017-04-05:
@lr-morales that is a good idiea I've just include a new option to enable/disable ROS compatibility ENABLE_ROS which is enabled by default. Also if the dependencies are not resolved ROS will be disabled anyway raising a warning during CMake checks.

aitormf, 2017-04-07:
#737 merged, now ROS is optional

closed 2017-04-07 by aitormf