JdeRobot/base #731: [Plugins] Add support for a bumper sensor in the turtlebot gazebo plugin.

fqez, 2017-03-22:
Our current plugin of the turtlebot has support for the following sensors:

* cameras
* laser
* pose3d

It would be a good idea to integrate a bumper sensor also.

fqez, 2017-03-22:
The ICE interface has to be improved in order to get simplier information from the sensor, like ```bumper->hadContact()``` or ```bumper->Active()```.
It now can be infered by the number of contacs read by the sensor with something like this:
```
int c = 0;
....
if (bumperData.numContacs >c) {
    std::cout << "we hit something!" << std::endl;
    c = bumperData.numContacts;
}
```
I count with the fast processing time between readings to detect a collision, but it would be more useful to has an encapsulated version of the code avobe or something more efficient.
Now the slice interface has information for the number of collisions and the name of the collided bodies only. It should give info about if the sensor is active when the function is called.

**This feature hasn't been tested yet because of problems with gazebo and my virtual machine, but it compiles and links properly**

aitormf, 2017-04-05:
merged #732

closed 2017-04-05 by aitormf