JdeRobot/base #740: CMake error when building TeachingRobotics

herozero777, 2017-03-30:
CMake Error at /usr/local/share/jderobot/Deps/python/Modules/ice_py/CMakeLists.txt:3 (usePython):
  Unknown CMake command "usePython".
Call Stack (most recent call first):
  /usr/local/share/jderobot/Deps/LoadJderobotDependencies.cmake:12 (include)
  cmake/Deps/jderobot-deps/CMakeLists.txt:14 (include)
  CMakeLists.txt:32 (include)

LujunWeng, 2017-03-30:
Actually, it does not need to be built. Just go to `src` and read README.md for instruction.

herozero777, 2017-03-31:
src in TeachingRobotics directory ?
Sorry i checked both the directories and I unable to find a README.md file that can help me run exercises.
Also can you please tell me where would be the practice folder located ?
Thank You

jmplaza, 2017-04-01:
I don't understand the issue either. TeachingRobotics repository (by the way it has been renamed to Academy) does not need to be built at all. They are Python JdeRobot components so there is no need to compile them. The src directory in Academy repository is a collection of robotics and computer-vision practices. You can see their [presentation and videos](http://jderobot.org/Teaching_robotics_with_JdeRobot)

Most directories there  (like global_navigation, follow_line....) have a README.md file with instructions to launch the practice. Unfortunately most of them are in Spanish (they will be translated to English soon), but I think you can follow them as the commands are in Linux :-) and the Python methods and variables in English.

closed 2017-04-01 by jmplaza