JdeRobot/base #792: Fix compilation warnings

chanfr, 2017-04-29:
The idea of this issues is to reference all the commits related to remove compilation warnings.

lr-morales, 2017-07-28:
Hello,

Regarding this issue, I have found lots of deprecation warnings, that could be listed in two groups:

- **Own** deprecations: Most of warnings in this group come from the use of [progeo](https://github.com/JdeRobot/JdeRobot/tree/master/src/libs/progeo).

- **External** / **Standard library** deprecations: These come from the usage of deprecated functions/classes/methods/... from external libraries or from the standard library.

About **own** ones, we should think about that if the tool using the library is also deprecated, in which case I would remove it from the master branch (maybe move them to another branch?) or if it is an active tool. In this last case, what is the `progeo` replacement?

About **standard library** ones, the most noticeable one is the deprecation of `std::auto_ptr`. I'm not sure if we are supporting C++ standards below C++11. If we aren't, the solution seems to use instead `std::unique_ptr` or `std:shared_ptr`. If we are, these aren't easily solvable, as the "correct" solution for each standard should be selected. 

The **external** ones could be reviewed after checking the other two.

fqez, 2017-07-28:
Hi @lr-morales you are right, most of our warnings come from deprecated code. 

I am not aware about which instructions are deprecated in the progeo library or if it is the whole library problem itself (maybe @chanfr or @jmplaza knows better). In any case we should at least review our two progeo copies: the one you linked and the one that is inside geometry lib. I don't know if there is a replacement for the progeo library at the moment, but we could take advantage of this opportunity and "remake" it in C++ removing all the deprecations in the process.

About the C++ standard warnings, I would have to review each one, but we should ideally be supporting only C++11 standards as we are currently using the latest stable gcc and g++ versions (I think it's 5.4.0 in Ubuntu Xenial).

What do you think?

lr-morales, 2017-07-28:
Confirming gcc versions of Xenial being 5.4.0 and Jessie being 4.9.2, both should support `--std=c++11` with `std::unique_ptr` and `std::shared_ptr`, so I agree with the idea of supporting the C++11 standard. I'll wait other opinions regarding this point before assuming that standards below this one will not be supported.

After reviewing the [progeo](https://github.com/JdeRobot/JdeRobot/tree/master/src/libs/geometry/progeo) of geometry (I didn't remember it was there, thanks for pointing it out), it seems to be already the C++ version of the one deprecated. If this is correct, then the point moves to see if the libraries/tools using the old `progeo` should be deprecated too and at least removed from the main compilation chain. Again I'll wait to @chanfr, @jmplaza or someone working more actively with progeo to confirm this, as you suggested.

lr-morales, 2017-08-01:
When the PR #886 is reviewed and merged, the remaining warnings should be (assuming gcc version in xenial):

- Our deprecated `progeo`.
- C++11 standard deprectated `std::auto_ptr`.
- Others, like
  - In `kinect::KinectPlugin::LoadSensors`, the call `gazebo::physics::Base::GetChild(0)` is reported as ambiguous (`unsigned int` / `const string&`).
  - `XnCppWrapper.h` has `#pragma warning`, which are reported as unknown by the compiler.

jmplaza, 2017-08-02:
Hi, 

good job @lr-morales with PR #886!

good to know the C++11 is supported on Jessie (gcc-4.9.2). Let's move forward jumping to stdC++11 and marking as deprecated (and so not-supported) previous releases.  

Yes C release of progeo library [is deprecated](https://github.com/JdeRobot/JdeRobot/tree/master/src/libs/progeo). It was marked as deprecated 4 months ago. It's time to remove it from the official code. The C++ progeo release is [here](https://github.com/JdeRobot/JdeRobot/tree/master/src/libs/geometry/progeo)

I don't know whether @eperdices uses this C++ progeo library for [SDVL](https://github.com/JdeRobot/slam-SDVL) or another code is used there. In addition I don't know whether @chanfr  uses this C++ progeo library for PeopleTracker application or another code is used there. There is code for similar functionality in OpenCV. Anyway AFAIK the C library can be safely removed.

closed 2021-02-21 by jmplaza