JdeRobot/base #831: New cars for gazebo

fqez, 2017-06-18:
The current vehicles of JdeRobot are a little messy and "unreal". We need to include new and more realistic looking cars in order to make the simulations more attractive.
It would be good to include kind of holonomic and non holonomic cars, meaning:

* "Holonomic" cars can rotate on its own axis like it has 360º wheel.
* "Non holonomic" cars behave like a real world car (ackermann geometry).

fqez, 2017-06-18:
For the holonomic cars, I've adopted an sketchup model of a Audi RSQ (from the movie i,Robot) like shown in the image below. I've selected this car because it has no real wheels (the car levitates), so it makes more credible the fact that it can turn around its own axis.
I've made a normal and a taxi version of the car applying some colors and stuff.

* [x] Holonomic taxi
* [x] Holonomic car
* [ ] Non holonomic taxi
* [ ] Non holonomic car

**Taxi version:**
[image]

**Normal version:**
[image]

This is how it looks in Gazebo:

**Both taxi and normal versions:**
[image]

For the "non holonomic" car, I could choose whichever car I want or like, so I made my mind for the Mercedes SLS AMG. Again, this model has its normal and taxi version.

**Taxi version:**
[image]

**Normal version:**
[image]

jmplaza, 2017-06-19:
Awesome Fran, fantastic cars! :-)

fqez, 2017-06-21:
Now both holonomic cars have their own plugin (just as temporary solution). It would be good to refactor also the JdeRobot's gazebo plugins because there is so much code duplicated.

closed 2021-02-21 by jmplaza