JdeRobot/base #864: [drivers] [kobuki] Kobuki dependencies are not met when ros-kinetic-kobuki-* packages are installed on Ubuntu 16.04

okanasik, 2017-07-04:
Kobuki dependencies are not met when ros-kinetic-kobuki-* packages are installed on Ubuntu 16.04. In the cmake configuration part, kobuki_LIBRARY11, kobuki_LIBRARY8, and kobuki_LIBRARY9 cannot be found and JdeRobot cannot be configured. Those libraries are ecl_statistics, ecl_ipc, and ecl_io. I could not find any package that installs those libraries. If we can remove those libraries from Deps/kobuki/CMakeLists.txt, we are able to configure and compile without any problem.

Is there any dependency for these libraries?

aitormf, 2017-07-05:
This problem occurs because Ros puts their libraries first in the LD_LIBRARY_PATH and when compiling this using ROS instead of ours.

Actually I don't know how solve this

aitormf, 2017-07-06:
Hi @okanasik, 
I've been looking, and with the following packages installed everything works for me:
<pre>
ros-kinetic-kobuki-auto-docking/xenial,now 0.7.4-0xenial-20170613-195912-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-bumper2pc/xenial,now 0.7.4-0xenial-20170613-175352-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-capabilities/xenial,now 0.7.4-0xenial-20170615-105222-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-description/xenial,now 0.7.4-0xenial-20170627-170511-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-dock-drive/xenial,now 0.7.8-1xenial-20170402-115048-0700 amd64 [instalado, automático]
ros-kinetic-kobuki-driver/xenial,now 0.7.8-1xenial-20170402-115122-0700 amd64 [instalado, automático]
ros-kinetic-kobuki-ftdi/xenial,now 0.7.8-1xenial-20170402-114959-0700 amd64 [instalado, automático]
ros-kinetic-kobuki-gazebo-plugins/xenial,now 0.5.7-0xenial-20170627-174132-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-keyop/xenial,now 0.7.4-0xenial-20170613-185658-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-msgs/xenial,now 0.7.0-0xenial-20170613-161112-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-node/xenial,now 0.7.4-0xenial-20170615-104621-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-random-walker/xenial,now 0.7.4-0xenial-20170613-185702-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-rapps/xenial,now 0.7.4-0xenial-20170613-204723-0800 amd64 [instalado, automático]
ros-kinetic-kobuki-safety-controller/xenial,now 0.7.4-0xenial-20170613-182445-0800 amd64 [instalado, automático]
</pre>

cmake kobuki output:
<pre>
kobuki was found in: /opt/ros/kinetic/include
kobuki LIBRARY FOUND IN /opt/ros/kinetic/lib/libkobuki.so;/opt/ros/kinetic/lib/libkobuki_dock_drive.so;/opt/ros/kinetic/lib/libecl_devices.so;/opt/ros/kinetic/lib/libecl_errors.so;/opt/ros/kinetic/lib/libecl_exceptions.so;/opt/ros/kinetic/lib/libecl_formatters.so;/opt/ros/kinetic/lib/libecl_geometry.so;/usr/local/lib/libecl_io.so;/usr/local/lib/libecl_ipc.so;/opt/ros/kinetic/lib/libecl_mobile_robot.so;/usr/local/lib/libecl_statistics.so;/opt/ros/kinetic/lib/libecl_streams.so;/opt/ros/kinetic/lib/libecl_threads.so;/opt/ros/kinetic/lib/libecl_time.so;/opt/ros/kinetic/lib/libecl_time_lite.so;/opt/ros/kinetic/lib/libecl_type_traits.so
</pre>

okanasik, 2017-07-06:
Yes, you are right. The missing libraries in the JdeRobot ThirdParty repository. First I installed sophus library https://github.com/JdeRobot/ThirdParty/tree/master/sophus and then kobukilib library https://github.com/JdeRobot/ThirdParty/tree/master/kobukilib

closed 2017-07-06 by okanasik