JdeRobot/base #872: Python interfaces install path locked at lib root location

lr-morales, 2017-07-08:
While checking new responses about #485,  found that JdeRobot install output reports multiple  `__init__.py` installs at python lib root location as this example:
 
    -- Installing: /usr/lib/python2.7/__init__.py
    -- Installing: /usr/lib/python2.7/__init__.py

this file contains Ice declarations after JdeRobot finishes installing. As all the ice interfaces are installed in this path too, seems that these declarations are needed; but maybe isn't a good idea having all these files located at that path like this (ej: breaking an existing __init__.py or some other package breaking ours). It could be a good idea thinking to move all these interfaces to some subpath or a common module, checking the implications of these in the python components that make use of the interfaces.

aitormf, 2017-07-17:
We need modify python ICE interfaces creation to solve this

aitormf, 2017-07-20:
with #880 this issue will be solved.

I have edited python install clause of interfaces to solve it

lr-morales, 2017-07-26:
Regarding the solution, and knowing that they are different issues, will this modification solve also #485 for python-related libraries?

closed 2017-07-27 by fqez