JdeRobot/base #882: [Tools] Colortuner enhancements needed

fqez, 2017-07-25:
In order to make this tool more usable, we may need to add some features such as:

   * Pixel color selector (with zoom mode)
   * Adjust the size of the interfaze and make it less intrusive.
   * Some visual adjustments, just to wash its face.

fqez, 2017-07-25:
After a couple of days of fighting with QT, I've managed to improve quite a bit the colorTuner tool.

Main changes are:
   * Reduce the size of the tool, now it is less intrusive for the user. This feature depends on the aspect ratio of the video that is showing. The default aspect ratio now is 16:9, and the image size is hardcoded as 640x320. It will be configurable throught a .cfg file in a next iteration
   * Added a pixel selector. Now if you move the mouse on the source image, the pixel beneath the cursor is selected and showed at the right of the window with its coordinates and RGB value.
   * Added a zoom feature. This enhancement helps to select pixels by re-scaling the source image in a factor (constant in the code, but will be configurable) and showing it at the right of the window.
   * The filter values now can be specified by a QLineEdit, typing them down instead of adjusting an slider. Sliders are kept also, so the user can chose which one he wants to use. This helps to fine tuning the filter values.
   * Added an about dialog. I thought that keeping a huge logo in the main window of the gui is ugly and not very usable, because it causes the GUI to grow unnecessarily, so I decided to add an "about" dialog with the logo and some interesting links: like the JdeRobot's main web or this github repository. This dialog can be shown clicking in the "FIle->about" menu of the menu bar of the tool.
   * Added a Dark theme to the tool. I've integrated an stylesheet from [here](https://github.com/ColinDuquesnoy/QDarkStyleSheet) because it looks cool and it's super easy to set in python. The good thing is that this theme is portable to C++ also. This needs to be integrated in each tool (makes possible an ad-hoc configuration) or in the JdeRobot libs.
   * Changed the architecture of the GUI. I've manually reconstructed the whole tool GUI, so it is more scalable and easier to modify by code. The following schema shows how the new ColorTuner is built:

[image]

And here are some pictures showing the current version of ColorTuner:

[image]

And the new one:

[image]

BONUS: new version with dark theme:

[image]

Here is a video showing the changes:

https://youtu.be/VNwjfzoCZ-I

jmplaza, 2017-07-26:
Awesome @fqez ! :-) 
Maybe 640x480 or 320x240 are more common sizes for real cameras....

commit 7079f673 (2017-10-09): Merge pull request #912 from fqez/issue_882_1

[fix #882] New version of colortuner
closed 2017-10-09 by aitormf