JdeRobot/base #888: [Tools] cameraview_py displays output in BGR color space

ni9elf, 2017-07-28:
The color space used by camerview_py is BGR instead of RGB. The following line in OpenCV will convert the color space to RGB and solve the issue:
`image = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)`

aitormf, 2017-07-28:
Hi @ni9elf, 
The problem is backwards, Imshow uses images in BGR and we give it in RGB, so it looks bad, we have to move from RGB to BGR instead of BGR to RGB (in practice, for this case, is the same because both swap channels R and B)
[source](https://stackoverflow.com/questions/38598118/difference-between-plt-show-and-cv2-imshow)

ni9elf, 2017-07-29:
Hi @aitormf 
Thanks for the feedback. Yes, RGB to BGR will be technically correct.

aitormf, 2017-07-31:
solved with #889

closed 2017-07-31 by aitormf