JdeRobot/base #893: Gazebo "Tutrlebot Jde" model issue

Voidminded, 2017-08-13:
I have an issue with **Turtlebot Jde** model in Gazebo. When I load any world containing this model (i.e. kobuki-simple, montmelo-kobuki) Gazebo crashes while initialising, right after initialising the laser. Doesn't matter if I'm running it headless (`gzserver`) or with the client (`gazebo`). 

Even if I open the Gazebo with an empty world, as soon as I add this model to the world Gazebo crashest.

Here is the output:

> ➜  ~ gazebo kobuki-simple.world
> TurtlebotSensors::TurtlebotSensors()
> [0x7f9e1403eaf0]TurtlebotPlugin::TurtlebotPlugin()
> cam_turtlebot_left
> cam_turtlebot_right
> camera
> laser
> left_cliff_sensor
> leftfront_cliff_sensor
> right_cliff_sensor
> rightfront_cliff_sensor
> wall_sensor
> [turtlebot] TurtlebotPlugin::InitializeIce()
> [turtlebot] 	config: turtlebotplugin.cfg
> [Info] loaded Ice.Config file: /home/sepehr/Dev/colab-gsoc2017-SepehrMohaimanian/build/install/share/jderobot/conf/turtlebotplugin.cfg
> [turtlebot] 	create Ice plugin...
> [turtlebot] TurtlebotPlugin::Init()
> [turtlebot] Sensors of turtlebot
> [turtlebot] 	cam_turtlebot_left (id: 673)
> [turtlebot] 	cam_turtlebot_right (id: 689)
> [turtlebot] 	laser (id: 114)
> ➜  ~       
> 

But it's working perfectly with every other models and worlds.

The exact same issue appeared once before, I spent a couple of hours to find the issue, but was not successful. I have already tried uninstalling and deleting everything, cloning, and recompiling the whole JdeRobot project from scratch a dozen of time, but it doesn't help. Last time I had the same issue I ended up reinstalling the Ubuntu ! which I really don't want to do again, as this keep showing up.

Does anyone have any idea what could be wrong ? My initial guess is an empty tag in the Turtlebot model or sth.

aitormf, 2017-08-13:
Hi @Voidminded, 
try to run gazebo with **--verbose** to see more information about crash

Voidminded, 2017-08-13:
Verbose mode doesn't show much extra info, here's the output:

Running the Gazebo:

> ➜  ~ gazebo --verbose kobuki-simple.world
> Gazebo multi-robot simulator, version 7.0.0
> Copyright (C) 2012-2016 Open Source Robotics Foundation.
> Released under the Apache 2 License.
> http://gazebosim.org
> 
> [Msg] Waiting for master.
> Gazebo multi-robot simulator, version 7.0.0
> Copyright (C) 2012-2016 Open Source Robotics Foundation.
> Released under the Apache 2 License.
> http://gazebosim.org
> 
> [Msg] Waiting for master.
> [Msg] Connected to gazebo master @ http://127.0.0.1:11345
> [Msg] Publicized address: 192.168.0.20
> [Msg] Connected to gazebo master @ http://127.0.0.1:11345
> [Msg] Publicized address: 192.168.0.20
> [Wrn] [msgs.cc:1655] Conversion of sensor type[depth] not suppported.
> TurtlebotSensors::TurtlebotSensors()
> [0x7fbd6403eb20]TurtlebotPlugin::TurtlebotPlugin()
> cam_turtlebot_left
> cam_turtlebot_right
> camera
> laser
> left_cliff_sensor
> leftfront_cliff_sensor
> right_cliff_sensor
> rightfront_cliff_sensor
> wall_sensor
> [Wrn] [turtlebotcontrol.cc:67] No torque value set for the DiffDrive plugin.
> [turtlebot] TurtlebotPlugin::InitializeIce()
> [turtlebot] 	config: turtlebotplugin.cfg
> [Info] loaded Ice.Config file: /home/sepehr/Dev/colab-gsoc2017-SepehrMohaimanian/build/install/share/jderobot/conf/turtlebotplugin.cfg
> [turtlebot] 	create Ice plugin...
> [turtlebot] TurtlebotPlugin::Init()
> [turtlebot] Sensors of turtlebot
> [turtlebot] 	cam_turtlebot_left (id: 673)
> [turtlebot] 	cam_turtlebot_right (id: 689)
> [turtlebot] 	laser (id: 114)
> ➜  ~ 

and running headless:

> ➜  ~ gzserver --verbose kobuki-simple.world
> Gazebo multi-robot simulator, version 7.0.0
> Copyright (C) 2012-2016 Open Source Robotics Foundation.
> Released under the Apache 2 License.
> http://gazebosim.org
> 
> [Msg] Waiting for master.
> [Msg] Connected to gazebo master @ http://127.0.0.1:11345
> [Msg] Publicized address: 192.168.0.20
> [Wrn] [msgs.cc:1655] Conversion of sensor type[depth] not suppported.
> TurtlebotSensors::TurtlebotSensors()
> [0x7fb2f403eaf0]TurtlebotPlugin::TurtlebotPlugin()
> cam_turtlebot_left
> cam_turtlebot_right
> camera
> laser
> left_cliff_sensor
> leftfront_cliff_sensor
> right_cliff_sensor
> rightfront_cliff_sensor
> wall_sensor
> [Wrn] [turtlebotcontrol.cc:67] No torque value set for the DiffDrive plugin.
> [turtlebot] TurtlebotPlugin::InitializeIce()
> [turtlebot] 	config: turtlebotplugin.cfg
> [Info] loaded Ice.Config file: /home/sepehr/Dev/colab-gsoc2017-SepehrMohaimanian/build/install/share/jderobot/conf/turtlebotplugin.cfg
> [turtlebot] 	create Ice plugin...
> [turtlebot] TurtlebotPlugin::Init()
> [turtlebot] Sensors of turtlebot
> [turtlebot] 	cam_turtlebot_left (id: 673)
> [turtlebot] 	cam_turtlebot_right (id: 689)
> [turtlebot] 	laser (id: 114)
> [1]    24741 segmentation fault (core dumped)  gzserver --verbose kobuki-simple.world
> ➜  ~ 
>

jmplaza, 2017-08-16:
Uhm... weird. @jorgesate also reported [the same problem](http://jderobot-developer-list.2315034.n4.nabble.com/JdeRobot-Academy-problems-running-3d-reconstruction-td4643326.html#a4643332) in a different example, but no solution was fount AFAIK.

I've the latest JdeRobot from debian package (5.5.2) and the [example involving the TurtleBot plugin](http://jderobot.org/index.php/Running_JdeRobot#Simulated_Kobuki_.2B_KobukiViewer) works fine... 

One way to circumvent the problem is to use the ROS plugin for the Kobuki robot, like in the [kobuki-simple-ros.world](https://github.com/JdeRobot/JdeRobot/blob/master/src/drivers/gazeboserver/worlds/kobuki-simple-ros.world). Using the jderobotComm library ICE-applications can connect now to ROS driver nodes...

Hope it helps

commit 5ba0a9d2 (2017-11-02): Merge pull request #943 from aitormf/issues/893

[fix #893] solved bug in turtlebotJde model
closed 2017-11-02 by fqez