JdeRobot/base #895: Confusion with config file for the uav_viewer_py tool

raulperula, 2017-08-17:
Hi!

Trying to launch the _uav_viewer_py_ tool from the [example page](http://jderobot.org/index.php/Running_JdeRobot#Aerial_robots), by the command:

`uav_viewer uav_viewer_simulated.cfg
`

I have noticed that the config file that is included in the [tool folder](https://github.com/JdeRobot/JdeRobot/tree/master/src/tools/uav_viewer_py) is different and does not work:

`uav_viewer uav_viewer_py.cfg
`

Is this an error or simply a "residual" file?

aitormf, 2017-08-18:
Hi @raulperula,

yes,  **uav_viewer_py.cfg** is a residual file. uav_viewer_py uses config files included in uav_viewer to reduce config files number

raulperula, 2017-08-18:
Ok! So should I remove that file or could it be done by someone that has direct access to the repository. I think a pull request only for removing a file may be unnecessary.

aitormf, 2017-10-04:
with new changes that allow us install uav_viewer_py without uav_viewer, we need a config file for uav_viewer_py. I will edit this file to fix it

commit 0fd066e3 (2017-10-04): [fix #895] updated uav_viewer_py.cfg

uav_viewer_py.cfg has a simulated config now
closed 2017-10-04 by aitormf