JdeRobot/base #896: jderobotcomm_py library does not support drone's commands

raulperula, 2017-08-17:
Hi!

I am trying to use the drone interfaces to control them but I have notice that there is no common interface implemented in the [library](https://github.com/JdeRobot/JdeRobot/tree/master/src/libs/jderobotcomm_py) yet.

Is it planned to be included?

At this moment, I am using directly the functions implemented in the _uav_viewer_py_ tool but adapted.

aitormf, 2017-08-18:
Hi @raulperula,

yes, it is planned to include drone command to jderobotcomm_py (#679). The only thing we thought we'd do next month. After changing the configuration files  (#892) to not have to be modifying so many files when making this change.

If you need it already, we can change the order and include commads first

raulperula, 2017-08-18:
No, it is ok for me. Once we have the new format for the config files I will update the Scratch2JdeRobot tool, and later, once the drone interface is added I simplify my code.

Thank you for all your comments @aitormf

PR #937 (2017-10-24): added drone interfaces to comm
fix #896
commit 93bbb77c (2017-10-24): Merge pull request #937 from aitormf/issues/896

fix #896  added drone interfaces to comm
closed 2017-10-24 by fqez