JdeRobot/base #897: jderobotcomm_py lib bug?

raulperula, 2017-08-17:
Hi!

I have noticed that in the code of the [jderobotcomm_py](https://github.com/JdeRobot/JdeRobot/blob/master/src/libs/jderobotcomm_py/jderobotComm/__init__.py) library the ROS node is returned:

```
...
	if (rosserver):
		node = rospy.init_node(nodeName, anonymous=True)

	return ic,node
...
```

But the [rospy API](http://docs.ros.org/kinetic/api/rospy/html/rospy-module.html#init_node) shows that there is nothing to return.

Does it has any sense to return it?

aitormf, 2017-08-18:
Hi @raulperula,
Not is a bug. This function is prepared by the time that it generates ice communitacor and ros node.
In this way, customers are modified as little as possible

raulperula, 2017-09-28:
Ok, thanks for the clarification. I think we can close the issue then.

closed 2017-09-28 by raulperula