JdeRobot/base #929: [lib] [config] Trying to access not existing keys gives KeyError

okanasik, 2017-10-14:
Trying to access not existing keys gives KeyError, but it should use the default value. In libs/comm_py it is coded to set the default value but config library gives KeyError. The following error text might be useful. The yml file does set maxW value, but config library gives KeyError
```
loading Config file ./sample1.yml
Publishing sample1.myMotors with ICE interfaces
Traceback (most recent call last):
  File "./sample1.py", line 103, in <module>
    interfaces = Interfaces()
  File "./sample1.py", line 56, in __init__
    self.connectProxies()
  File "./sample1.py", line 61, in connectProxies
    self.myMotors = self.jdrc.getMotorsClient("sample1.myMotors")
  File "/opt/jderobot/lib/python2.7/comm/communicator.py", line 85, in getMotorsClient
    return getMotorsClient(self, name)
  File "/opt/jderobot/lib/python2.7/comm/motorsClient.py", line 99, in getMotorsClient
    return cons[server](jdrc, prefix)
  File "/opt/jderobot/lib/python2.7/comm/motorsClient.py", line 23, in __getMotorsIceClient
    client = MotorsIceClient(jdrc, prefix)
  File "/opt/jderobot/lib/python2.7/comm/ice/motorsIceClient.py", line 46, in __init__
    maxWstr = jdrc.getConfig().getProperty(prefix+".maxW")
  File "/opt/jderobot/lib/python2.7/config/properties.py", line 34, in getProperty
    return self._searchNode(self._config, names)
  File "/opt/jderobot/lib/python2.7/config/properties.py", line 51, in _searchNode
    return (self._searchNode(nod, lst))
  File "/opt/jderobot/lib/python2.7/config/properties.py", line 51, in _searchNode
    return (self._searchNode(nod, lst))
  File "/opt/jderobot/lib/python2.7/config/properties.py", line 48, in _searchNode
    nod = node[name]
KeyError: 'maxW'
```

aitormf, 2017-10-16:
Hi @okanasik, 
can you show me sample1.yml content?

okanasik, 2017-10-16:
Hi @aitormf 
Here is the content of sample1.yml:
```
sample1:
  NodeName: sample1
  myMotors:
    Name: myMotors
    Proxy: Motors:default -h localhost -p 9001
    Server: 1
    Topic: ''
```

aitormf, 2017-10-17:
It is a mistake the Comm library, I thought I read the property with "getPropertyWithDefault" so that it is not obligatory to have those values in the configuration file but not, so it shows this error

PR #934 (2017-10-20): [issue #929] added default values to some config to comm_py
fix #929
commit 18d08f1a (2017-10-20): Merge pull request #934 from aitormf/libs/comm_py/929

[fix #929] added default values to some config to comm_py
closed 2017-10-20 by fqez