JdeRobot/base #945: [tools] Scratch2JdeRobot new web and examples

jmplaza, 2017-11-04:
A page in the Tools section of JdeRobot webpage describing how to use Scratch2JdeRobot would help to new users who start using this tool.

In addition, one or two small behaviors already programmed as example would clarify even more the use of Scratch2JdeRobot. They can be included in the src/examples/Scratch2JdeRobot directory, for instance, and a brief recipe in the examples section of JdeRobot wepage. One with TurtleBot robot in Gazebo and another with a drone, as both are already supported.

What do you think @raulperula  @lr-morales?

The initial release was developed by @raulperula @lr-morales and @jmplaza in [this project](http://jderobot.org/Raulperula-colab) inside Google Summer of Code 2017. It is not fully and smoothly integrated in current JdeRobot release yet, but we are working on that. In addition, @scarrionv is also working on improving this tool. In [this link](http://jderobot.org/Scarrion-tfg) some recent videos are available of programming a simulated TurtleBot with Scratch.

raulperula, 2017-11-07:
I have added a basic TOC. Please tell me if you both agree or if there is something you miss.

scarrionv, 2017-11-07:
That's perfec. I already have a few steps for "How to create a New block" in my wiki (http://jderobot.org/Scarrion-tfg) if you want to use it.

jmplaza, 2017-11-08:
I've reorganized the Scratch2JdeRobot web a little bit, aimed to _users_ of this tool more than in _developers_.  Some of the blocks description are now wishful (we are still working on them) but @scarrionv @raulperula Please feel free to improve it further, in particular filling the directions to load an example program with the tool and run it with the simulated robot.

scarrionv, 2017-11-08:
I've added some new content to the wiki, take a look when you can

jmplaza, 2017-11-08:
Good job @scarrionv . We can update the links when a new PullRequest for #946 and #945 was merged

raulperula, 2017-11-09:
Nice structure @jmplaza  and good job @scarrionv . Take care with the wiki syntax, there is some list not well formated.

closed 2018-02-21 by fqez