JdeRobot/base #951: [drivers] MAVLink server for 3DR Solo Drone should be ported to python 2.7, instead of 3.5

jmplaza, 2017-11-12:
In order to be fully compatible with ROS Kinetic in JdeRobot we keep the code at Python 2.7. @Diegojnb is [this video](https://www.youtube.com/watch?v=zZa8ly6FBi8) with the 2.7 release of this driver or with the 3.5 one.

jmplaza, 2017-11-13:
In addition it should be changed to accept YML configuration files instead of old-fashioned, ICE style, .cfg

Diegojnb, 2018-02-11:
I have been looking for the dependencies that apply to this driver and are only necessary to have installed pyserial and pymavlink, both can be installed using: pip install (pyserial or pymavlink). 

Currently my installed versions are pyserial = 3.0.1 and pymavlink = 2.2.8. 

It would be necessary to make a debian package but I do not know how they are currently done and I would like you to help me out.

Thanks.

closed 2018-02-21 by fqez