JdeRobot/base #987: colorTuner error

vinay0410, 2017-12-12:
Ubuntu 16.04 LTS
Installation done from source code

Error says,
```ImportError: No module named jderobot```

whereas camera server is working fine

fqez, 2017-12-18:
Hi  @vinay0410 sorry for the late response.

It works well for me installing it from source code. Are you sure you have all the dependencies installed? Also, how did you install it, and how are you trying to run it?

try going to ```/opt/jderobot/share/jderobot/python/colorTuner_py``` and launch the component as follows:

```python colorTuner.py colorTuner_py.yml```

after you launch cameraserver, of course.

Make sure you got the latest version of the repo, also.

Hope it helps!

vinay0410, 2017-12-20:
Well, Thank You, for the reply. I deleted everything including dependencies and did a clean install again.
And it worked!

closed 2017-12-20 by vinay0410