JdeRobot/base #1000: [Tools] RGBDViewer is not using the comm library.

fqez, 2017-12-19:
This tool is currently working with easyice instead of comm. A refactorization is needed.

vinay0410, 2017-12-20:
I would like to work on that, if you are fine with that.
It would be very helpful, if you could help me with a head start amor guidelines

fqez, 2017-12-20:
Of course @vinay0410! feel free to attack any issue you want, we are glad to got some help :)

You can start by reviewing some other components already ported to use comm instead of easyIce.
I would recommend you to start by checking [this one](https://github.com/JdeRobot/JdeRobot/tree/master/src/examples/basic_component) out. Also, as this tool works mainly with cameras, you may want to check out [this one](https://github.com/JdeRobot/JdeRobot/tree/master/src/tools/cameraview) too.

The main idea here is to make the component independent from the communications layer, so it must have no ICE nor ROS dependencies/links/references in it. Also, the way the configuration files are managed should be replaced by the new format in YAML (yml format).

You should refactor at least the CMakeLists.txt, rgbdViewer.{cpp,h}, rgbdViewer.cfg. 

## ICE

```
#include <Ice/Ice.h>
#include <IceUtil/IceUtil.h>
#include "parallelIce/cameraClient.h"
#include "parallelIce/pointcloudClient.h"

#include "easyiceconfig/EasyIce.h" 

[...]

try{
	ic = EasyIce::initialize(argc,argv);
	prop = ic->getProperties();
}catch (const Ice::Exception& ex) {
	std::cerr << ex << std::endl;
	return 1;
}

[...]

camRGB = jderobot::CameraClientPtr(new jderobot::cameraClient(ic, "rgbdViewer.CameraRGB."));
if (camRGB != NULL) {
	camRGB->start();
} else {
	throw "rgbdViewer: failed to load RGB Camera";
}

[...]

```

## JdeRobotComm

```
#include <jderobot/config/config.h> 
#include <jderobot/comm/communicator.hpp>
#include <jderobot/comm/cameraClient.hpp>
#include <jderobot/types/image.h>

[...]

Comm::CameraClient* camRGB;

Config::Properties cfg = Config::load(argc, argv);
Comm::Communicator* jdrc = new Comm::Communicator(cfg);

[...]

camRGB = Comm::getCameraClient(jdrc, "rgbdViewer.CameraRGB");

[...]
```

It is possible that there's a lack of some of the clients, like the pointCloud one. You can comment out that section if you don't feel like coding down the support for that client in the comm library.

Hope this is enough for you to start, let us know if you face any problem at all.

Thank you!!

vinay0410, 2017-12-22:
Thanks for the reply, and sorry for late reply. But actually I do have a Kinect and I was pretesting it so that I can validate my changes. But it fails to open without nay changes being done i.e with the original library.
Openni server says timeout, and I have tried many times, whereas its working fine with libfreenect.

fqez, 2017-12-22:
Hi @vinay0410 can you please share a screenshot showing your problem? Also, how are you launching the example?

vinay0410, 2017-12-22:
openniServer openniServer.cfg, and I get the following result.
[image]

fqez, 2017-12-22:
Yes, that trace is kinda "usual". Did you try to launch rgbdviewer after openniServer?

vinay0410, 2017-12-23:
I did that, and I got the following output
[image]
No viewer window was opened

fqez, 2017-12-25:
Ok @aitormf can you reproduce this? I don't have any RGBD sensor here, so it would be useful if you can confirm this issue. Thanks a lot!

aitormf, 2018-03-14:
solved

closed 2018-03-14 by aitormf