JdeRobot/base #999: [Server] YoutubeServer is not working properly with python2.7

fqez, 2017-12-19:
YoutubeServer works fine with python3, but it's incompatible with python2.7 right now.
It needs to be refactored and it also needs a CMakeLists.txt file for its installation.

fqez, 2017-12-19:
The PR #1001 solves the compatibility with youtubeserver and python2.7
Also I've modified it to allow the user to personalize a little bit more the component. It now offers:

* output path: where do you want the temporary files to be stored
* Video download format: in which format do you want the video to be downloaded
     if you want to know the available formats you only have to type: ```youtube-dl --list-formats [Youtube video URL]```
* It also let you choose the framerate in which ffmpeg will process the video.
* Now you can terminate the youtubeserver process just pressing Ctrl+c on the keyboard.

TODO:

It rest to include the necessary dependencies into the JdeRobot deps, and include the information to generate a debian package for this server. 

[Here is a video](https://www.youtube.com/watch?v=NwcndDBD3E0) showing how the server works

aitormf, 2017-12-20:
solved with #1001

closed 2017-12-20 by aitormf