JdeRobot/base #1003: New driver: picameraserver

aitormf, 2017-12-20:
we need make a new driver named picameraser to serve images from picamera (camera for Raspberry PI)

vinay0410, 2018-02-02:
Hi,
I have made a driver which can fetch images from Pi Camera using mmal (Multimedia Abstraction Layer ) and RaspiVid_State Struct [here](https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/raspicam).
But, I had a question.
Do you want to make this driver analogous to camera server or are you planning something else.

jmplaza, 2018-02-02:
Hi @vinay0410, 

nice! sure, that would be great. @jmvega has just finished the first release of this driver, check this [http://jderobot.org/JulioVega_PhD#2018.01.27._New_PiCam_ICE-driver_developed](video). 
For sure we can accept your C++ picamserver and test it. Accessing to camera parameters is a good new functionality. In addition we could compare the speed of both drivers in terms of framerate. 

Does it have a ICE (or ROS) wrapper to let other JdeRobot nodes connect to it and get the images?

Good job!

vinay0410, 2018-02-02:
I am using ICE, as used in cameraserver. Also, I have tested fetching images on raspberry pi, but to test jderobot build on raspberry pi, I have to install jderobot on my pi.
So, do I need to erase & install ubuntu 16.04 on my pi, or raspbian will do just fine ?

jmvega, 2018-02-02:
Hi Vinay,

  you can install jderobot on your pi following the steps described here [1]. But also, you can install only selected components (e.g. cameraserver and cameraview).

  I work with Raspbian and JdeRobot works fine. You can try this one or even Ubuntu Mate (Ubuntu for ARM systems).

  The actual picamserver issued in #1003 has been coded in Python and it also supports ICE protocol. It's nice to have the C++ version by your side!

  If you have any problem during the JdeRobot installation, do not hesitate, just ask.

  Julio

  [1] http://jderobot.org/index.php?title=Manual-5&oldid=19343#Installation_of_JdeRobot_in_ARM

vinay0410, 2018-02-03:
Thanks for the reply!
I was installing Jderobot on my pi from the link mentioned, but apt couldn't find some packages even after update, like all ros-kinetic packages.
Also, since I am using ICE, so I went on to install ICE which resulted in make error
```fatal error: lmdb.h: No such file or directory
 #include <lmdb.h>
```

It would be very helpful if you could guide me on installing, or if it helps, I can install Ubuntu 16.04 on my pi.

jmplaza, 2018-02-03:
The installation on Pi is kind of painful... Maybe @jmvega or @aitormf can share the binary SD image they created, putting it somewhere, and then you could download it and directly use it on your Pi.. What do you think?

vinay0410, 2018-02-03:
Yes, That sounds great, lets do that!

jmvega, 2018-02-08:
Hi Vinay,

  while we are extracting the Raspbian disk image and, especially,
uploading it (it's a 16GB MicroSD), let me know where are you stuck in
order to help you.

  Bests,

  Julio

vinay0410, 2018-02-08:
Hi Julio,
Actually, I'm almost done with the coding part. I have compiled it using cmake and also ran to see image from my pi camera, and is working great.
I have also made picameraserver using ice to test, but I'm not able to due to problems with jderobot installation.

So, in the meantime, if I install ubuntu server 16.04 on my pi to test and install jderobot, would it be worth it?

Thanks

jmplaza, 2018-02-09:
Hi @vinay0410,

 the SD image is HUGE (16GB), you can get it from [here](http://gsyc.es/jmplaza/jderobot-raspbian-disk_image.img). There is no need to install the whole JdeRobot middleware on your PI, just a few things. Maybe you can install only the required parts: ICE interfaces, cameraserver... from the source code.

Take a look at [this video](https://www.youtube.com/watch?v=hViz05u64IA), maybe you can record a new video from your PI including (a) cameraview connected to your mmal-based image server and (b) @jmvega 's picamserver.

What do you think?

vinay0410, 2018-02-11:
Sir, I tried to burn this image but is too large for a 16gb memory card, using a block size of 4M. If you haven't expanded your file system on the Pi, you could reduce ext4 partition size which would reduce image size so as to burn it in a 16 GB micro SD card.

Also, I tried to install jderobot on Ubuntu Mate on my pi. ICE 3.6 was causing test errors, therefore I tried 3.7 and was installed successfully after altering cmake files and ENV paths.
But I couldn't install jderobot due to a cmake error I couldn't fix, which is mentioned below:
```CMake Error: Couldn't determine link language for jderobot interfaces.```

I would also like to appreciate, the time & effort your organization is putting to help me install jderobot on my pi and would like to thank you all for the same.

Also, If it's a problem, then I could submit a pull request using the technique used by Cameraserver, so that minimal error occurs, and you can test it on your pi instead.

jmvega, 2018-02-11:
> Hi @jmvega <https://github.com/jmvega>,
> I have installed the image you've sent, but there are now jderobot header
> files present in it, only cameraserver_py and cameraview_py are present.
> Only the python module of jderobot is present

Hi Vinay,

  yes, we didn't install jderobot completely because of lots of
dependencies we found, but just these selected components. We installed
also OpenCV and Ice libraries, which probably you'll need.

  Do you need anything else?

  Julio

vinay0410, 2018-02-11:
Hi Julio,
I was thinking of using jderobotutil/interfacehandlers like cameratask.h and cameraHandle.h, I was using [CreateCustomImage)(https://github.com/JdeRobot/JdeRobot/blob/master/src/drivers/cameraserver/cameraserver.cpp) to fetch image from the camera using mmal and then publishing.

Please let me know if there is a better way to publish.

fqez, 2018-05-24:
The C++ version of this driver is pending for testing, I reopen this issue

commit bcbf3ec3 (2018-05-24): Merge pull request #1180 from jmvega/add_driver_piCamServer_py

[fix #1003] Added driver piCamServer_py to serve PiCamera and USB cams
closed 2021-02-21 by jmplaza