JdeRobot/base #1013: Improving ColorTuner to show color disc in GUI

jmplaza, 2017-12-26:
In the color disc an histogram of the current image should be shown too. This allows a better profiling of the color filters. 
And a visual definition of HSV color filters, including in a natural way color areas near the angular discontinuity (from instance from 350º to 10º in the color disc). We had this functionality in JdeRobot 4.3. release...

[image]

dattadebrup, 2018-02-09:
Hi,
I guess I will be able to implement the HSV colour disc feature as described [here](https://github.com/JdeRobot/JdeRobot/blob/4.3/schemas/hsvtuner/README). Should I work on it??

jmplaza, 2018-02-09:
Sure, feel free to submit your Pull-Request improving the current ColorTuner tool, in python. That would be a nice contribution.

dattadebrup, 2018-02-11:
Hi @jmplaza ,
I was working on this and I am able to make this much progress..(https://i.imgur.com/a0DFGMT.gif)
Right now I have to work on the bug where the arc is not getting drawn properly(as you can see in the gif) and then to integrate with the colortuner tool of Jderobot.Will it be alright if I submit my Pull request after 2 weeks as I have my college mid-term exams due in next week?
P.S.-I am Gsoc aspirant from India hoping to work under Jderobot.

pushkalkatara, 2018-02-17:
@dattadebrup Are you working on this issue?

dattadebrup, 2018-02-17:
Yes, I am...

fqez, 2018-02-19:
Wow @dattadebrup very nice job! I can't wait to get this enhancement integrated into colorTuner! Looking very nice, keep it up!

dattadebrup, 2018-02-20:
Thanks a lot @fqez for the appreciation.I myself can't wait to continue my work on this as soon as my exams end on Friday.

dattadebrup, 2018-02-26:
Hi,
I successfully made the HSV color range picker.
[image]
But I was having a hard time while trying to integrate it with the colortuner tool. Here is the [code](https://gist.github.com/dattadebrup/31491a8205b056f7c821015857a68213) of the standalone python file and the color disc image [file](https://i.imgur.com/wkQ8j9W.png) (rename it as wheelfinal.png after downloading the image) required to be kept in the same directory as the .py file. Can anyone please guide me how to integrate with the colortuner tool?

dattadebrup, 2018-03-01:
Hi,
I managed to integrate it with the colorTuner tool succesfully. Please review the PR. #1091

aitormf, 2018-03-19:
solved #1091

closed 2018-03-19 by aitormf