JdeRobot/base #1020: Extend Comm to support pointclouds

jmplaza, 2018-01-04:


pushkalkatara, 2018-02-20:
Hi @jmplaza . Please help me with the type of PointCloud structure which is required. There are two types of PointCloud messages in ROS - 
[PointCloud](http://docs.ros.org/api/sensor_msgs/html/msg/PointCloud.html)
[PointCloud2](http://docs.ros.org/api/sensor_msgs/html/msg/PointCloud2.html)

jmplaza, 2018-02-24:
Hi @pushkalkatara , for sure it will not be PointCloud as it is deprecated. According to [this thread](http://library.isr.ist.utl.pt/docs/roswiki/pcl(2f)Overview.html) maybe sensor_msgs::PointCloud2 is the winner as can be translated automatically into pcl::PointCloud<T> objects in the suscriber. The subscribers will work with pcl::PointCloud<T> objects as it seems to be the standard library for point clouds. Right? 

Do you find a way to directly publish pcl::PointCloud<T> objects? Maybe this could be the most efficient way.

> When subscribing to a pcl::PointCloud<T> topic with a sensor_msgs::PointCloud2 subscriber or viceversa, the conversion (deserialization) between the two types sensor_msgs::PointCloud2 and pcl::PointCloud<T> is done on the fly by the subscriber.

In addition, please test this [new example](http://jderobot.org/index.php/Examples#Kinect1_.2B_ROS_driver_.2B_RGBDViewer_tool). The idea is to extend the current comm library to support also "ROS pointclouds" and test it with RGBDViewer-tool using the extended comm library.

pushkalkatara, 2018-02-25:
ROS provides a package to directly convert [PointCloud to PointCloud2](https://answers.ros.org/question/136916/conversion-from-sensor_msgspointcloud2-to-pclpointcloudt/). 
So a new Type based on PointCloud2 and a [C++ and Python client](http://pointclouds.org/documentation/tutorials/) would I think to be a good approach for PointCloud2 support in the stack. 
Also, to test RGBD Viewer, Kinect is required. But I tried to use a [Rosbag](https://cs.nyu.edu/~silberman/datasets/nyu_depth_v2.html) file on the loop which contains both the topics needed. 
```
/camera/depth/image_raw
/camera/rgb/image_color
```
But while launching rgbdViewer, I faced this error - 
[image]
It says .world file not found which is traced [here](https://github.com/JdeRobot/JdeRobot/blob/9a34beb93e2af60c48968fe55f6508ba84833583/src/tools/rgbdViewer/drawarea.cpp#L651). How to load this file?

closed 2021-02-21 by jmplaza