JdeRobot/base #1044: CameraServer in JS using web technologies and ROS messages

jmplaza, 2018-01-29:
On Windows machines JdeRobot does not have yet any video source, only on Linux machines. Using the real camera from docker containers is difficult or impossible, but web technologies provide a (multiplatform) way for accessing video frames from real cameras. 
This new server will take advantage of such feature to provide images to any JdeRobot computer vision application running on Windows or MacOS. It will deliver them using the ROS communication layer.

pushkalkatara, 2018-02-15:
Hi. @jmplaza I would like to work on the issue. My approach would be setting up a NodeJS server based on roslibjs on the client side i.e on Windows or Mac. This Nodejs server would communicate to ROS through rosbridge_server on the port 9090 on which the roswebsocket is hosted.

jmplaza, 2018-02-15:
Hi @pushkalkatara, this issue is almost solved by @rperezgonz as you can see in [this video](https://www.youtube.com/watch?v=H3RvokYPsv4) at the JdeRobot's YouTube channel.... 

Some betatesting on the webtools would be useful anyway. We are about to update kobukiviewerjs, cameraviewjs, uavviewerjs [tools](https://github.com/JdeRobot/JdeRobot/tree/master/src/tools)

pushkalkatara, 2018-02-15:
I am up for betatesting, please guide me how to proceed?

jmplaza, 2018-02-24:
Hi @pushkalkatara , this week many of the PullRequests from @rperezgonz will be merged including the cameraserver-web and some webtools (uaviewer-web, cameraviewer-web...). Check [this video](https://www.youtube.com/watch?v=2BID7Zw1Ntc) for instance. Be aware and follow the directions we will update into the [examples page](http://jderobot.org/index.php/Examples) and report the testing on [Answers](http://jderobot.org/index.php/Examples)

pushkalkatara, 2018-02-25:
Sure thanks. @jmplaza