JdeRobot/base #1060: [Libs] YML config file description should be more readable

fqez, 2018-02-09:
The corrent implemtation of the comm library accepts Server value as 1 for ice and 2 for ros communications. It may be a good idea to make those values more understandable:

from: 

          Server: 1
           or
          Server: 2

to:

          Server: ice
           or
          Server: ros

pushkalkatara, 2018-02-19:
Hi @fqez I would like to work on this issue. My approach would be-

- Update file [ardroneextraClient.cpp - Line 14 ](https://github.com/JdeRobot/JdeRobot/blob/6998f05165c11d9df9f484bcb8b12e98b0b87b8c/src/libs/comm_cpp/src/ardroneextraClient.cpp#L14) to asStringWithDefault() function to receive a string.
 Further if string is "ice", make server = 1 else if string is "ros", server = 2. 
Similarly making the same changes in all related files located here [https://github.com/JdeRobot/JdeRobot/tree/master/src/libs/comm_cpp/src](Link)  

- Update file [Python Client - Line 74](https://github.com/JdeRobot/JdeRobot/blob/6998f05165c11d9df9f484bcb8b12e98b0b87b8c/src/libs/comm_py/comm/ardroneextraClient.py.only-ice.in#L74) check for "ice" or "ros" and change the value accordingly.

fqez, 2018-02-19:
Hi @pushkalkatara of course, feel free to dive into whatever issue you feel comfortable with!
I think that's a good approach, so go on.
Don't forget that this issue includes the task to modify every configuration file in the project, as well as both c++ and python version of every interface in comm.
If you face any problem, let us know.

Good luck!

pushkalkatara, 2018-02-20:
@fqez Please review the Pull Request whenever you are free.

commit f137a58a (2018-02-22): Merge pull request #1070 from pushkalkatara/feature-lib

[Libs] Makes YML config file description more readable fix #1060
closed 2018-02-22 by fqez