JdeRobot/base #1065: CMake Error: IceUtil not found.

pushkalkatara, 2018-02-15:
Deps/ice's CMakeLists.txt
```
#ICE installs c++11 libraries in this directory and we don't indicate this, cmake uses c++98 files
SET(CMAKE_PREFIX_PATH /usr/lib/x86_64-linux-gnu/c++11)
```
But Ice install binary files in the path 
```
/usr/lib/x86_64-linux-gnu
```
Hence dependency check is failing.

Also from Ice Documentation - 
The Ice 3.6 binary distributions for Linux include libraries with a ++11 suffix, such as libIce++11.so.36, and a c++11 subdirectory in lib or lib64 with symbolic links to these ++11 libraries. These libraries correspond to a build of Ice using GCC with --std=c++11 turned on. This "C++11 build" of Ice 3.6 provides some C++11 extensions available only in C++11 mode (see above).

closed 2018-02-17 by pushkalkatara

sprtkd, 2018-02-18:
How to resolve iceutil not found?

abhijayghildyal, 2019-03-06:
Try this:

https://ubuntu.pkgs.org/16.04/ubuntu-universe-i386/zeroc-ice35_3.5.1-6.4ubuntu1_all.deb.html

It worked for me ( :