JdeRobot/base #1068: raspberry pi installation instructions are outdated

vinay0410, 2018-02-19:
I would be altering Raspberry pi installation instructions and this is to keep a log of the updates, and will be mentioning changes in the thread

vinay0410, 2018-02-27:
Changes to do to build Zeroc Ice:
CPP11=yes flag doesn't work and actually activates concatenates --std=c++0x as noted in [Make.rules.Linux](https://github.com/zeroc-ice/ice/blob/3.6/cpp/config/Make.rules.Linux#L48).

So, it is necessary to edit this file and replace -std=c++0x by -std=c++11 in Line 49.

vinay0410, 2018-02-27:
Second:
When building Ice from source, path of it's executables is not added in Environment's path variable. Hence, commands like slice2cpp, slice2py, etc couldn't be run.
Cmake configures, but make fails giving an error ```Couldn't determine link language in jderobotinterface```.
Though I have submitted a pull request to cause a Fatal error for the same.
So I will be adding code blocks to add Ice's executables in path.

vinay0410, 2018-02-27:
Third:
Also for the error: ```Couldn't find header file lmdb.h```
It is necessary to install: ```sudo apt-get install libgflags-dev libgoogle-glog-dev liblmdb-dev```

Sai-Jiang, 2018-04-28:
As a supplement to the second step, some modifications to the CMakeList.txt are necessary for me to make the Ice library visiable.

INCLUDE_DIRECTORIES(/opt/Ice-3.6.4/include)
LINK_DIRECTORIES(/opt/Ice-3.6.4/lib/arm-linux-gnueabihf/c++11)
set (CMAKE_LIBRARY_PATH /opt/jderobot/lib/openni2 /opt/Ice-3.6.4/lib/arm-linux-gnueabihi/c++11)

closed 2021-02-21 by jmplaza