JdeRobot/base #1076: [Tools][VisualStates] Edit config files

reysam93, 2018-02-22:
Now, when creating the configuration file using Visual States is possible to add a new interface or to remove an existent one. It would be useful to edit the existing interfaces also, allowing to modify its fields and save it. Right now, the only way to do this is create a new interface with the desired fields and remove the old one.

pushkalkatara, 2018-02-22:
Hi @reysam93 . I would like to work on the issue.  My approach would be -

- [] Create new QPushButton "Edit" at column8 beside delete button.
- [] Create new function editClicked() which would open QInputDialog to take new config data.
- [] Update the configData and the QLabel.

jmplaza, 2018-03-14:
This will be solved on the VisualStates [new repository](https://github.com/JdeRobot/VisualStates/issues/5)

closed 2018-03-14 by jmplaza