JdeRobot/base #1112: [Gazebo] New world with mobile parts

fqez, 2018-03-23:
It's a good a idea to add a simulated world in gazebo emulating a gymkhana for drones in order to make the drone porgramming challenge (program-A-robot).

abhayraw1, 2018-03-23:
So, the mobile parts or obstacles would have a predefined trajectory or random?

fqez, 2018-03-23:
Hi there's already a first approach of this issue in the gymkhana branch of the repo, but it does need another iteration fixing:

* Collision with the moving walls, they should have more mass and an inertia matrix defined (default matrix is fine tho)

* Collisions with the hangar walls and roof, because it only has collision with the ground

* Moving walls need different color and (if possible) textures. Current colors are difficult to distinguish with the environment.

Here's a video showing the current state of the branch: https://www.youtube.com/watch?v=c9y89vZVkjg

Feel free to modify whatever you consider necessary :)

abhayraw1, 2018-03-23:
Thanks, I'll give it a shot.

fqez, 2018-03-23:
About the trajectory I would make it random to make this more fun. The current implementation consists on a dummy plugin for each wall moving them linearly. Let me know if you face any problem with this! good luck!

abhayraw1, 2018-03-24:
Hey, is the branch updated, because I see that `gazebo.world` only includes models for the hanger and the sun. 
We must also include a drone also, right?
Also I cant seem to see any inertial matrix defined.

fqez, 2018-03-24:
Hi, no there's no drone included in the world, I added it manually to record the video, but yeah, you can include it in the file with your changes! (its name is ArDrone2). About the inertial matrix, they are not included in the current version, that's why this issue needs another iteration.

abhayraw1, 2018-03-26:
No issues. I had already added those file. Was going to push them in the update. 

Is it okay if i change the meshes for the walls to some modified version of a basic cube?

commit 794e49f3 (2018-04-20): Merge pull request #1127 from JdeRobot/gymkhana

[fix #1112] Added new gymkhana world for drone contest
closed 2018-04-20 by aitormf