JdeRobot/base #1116: [Drivers] Refactor C++ CameraServer to use pure opencv

fqez, 2018-03-23:
The current implementation of the c++ cameraserver does not support image compression with the typical image formats (jpeg, png, etc). It is currently working with raw images, so this leads on a poor performance: low fps rate when sending images with ice. This is because raw images at 640x480 sent throught a socket from a webcam/video file/youtubeserver without any kind of compression make things slower.
The objective here is to reach at least 30fps sending images from the cameraserver to the cameraclient on localhost first. It would be nice also to support higher quality images (bigger image resolutions) without losing fps.

abhayraw1, 2018-04-05:
Maybe we could add a method in the [JdeRObot/image](https://github.com/JdeRobot/JdeRobot/blob/master/src/types/python/jderobotTypes/image.py) class to compress images and then use the inbuilt cv2 method in the corresponding receivers. 
What do you say @fqez ?

pushkalkatara, 2018-04-07:
I think we can update the [ThreadImage.py](https://github.com/JdeRobot/JdeRobot/blob/master/src/drivers/cameraserver_py/threadImage.py) script to preprocess the image for particular configurations and compression. ROS has specialized message CompressedImage for easy image transport for which it converts the Jpeg, PNG to [BGR8](https://github.com/ros-perception/image_transport_plugins/blob/indigo-devel/compressed_image_transport/src/compressed_publisher.cpp) and encodes the image using OpenCV's function. maybe we could similarly compress the image in ThreadImage.py and update the client to decode in the similar format.

commit b1373f26 (2018-04-18): Merge pull request #1126 from aitormf/drivers/cameraserver_py/ROS

[fix #1116 fix #1095] CameraServer_py is full ros and full opencv
closed 2018-04-18 by fqez