JdeRobot/base #1174: CamServerWeb and CamVizWeb don't get images from camera

PabloMorenoVera, 2018-05-21:
I'm trying to run this practice and i run the instructions. When i run CamServerWeb, a window appears and the configuration it's OK. The LED of my camera turns on.
When i launch CamVizWeb, the second windows appears and the configurations it's also OK.
But when i "Start" CamVizWeb, the image doesn't show.

I attach several images from the terminals.

- CamServerWeb connected:
[image]

-CamVizWeb connected:
[image]

-Starting CamViWeb:
[image]

- CamVizWeb image:
[image]

rperezgonz, 2018-05-22:
Hi,
Can you show me the developer tools? To open this, you can do it going to View menu in the top screen and it's opened a submenu. In this submenu, select Toggle Developer Tools.

It must be something like this:

[image]

And the configuration must be something like following:

[image]

PabloMorenoVera, 2018-05-23:
Hi @rperezgonz ,
I attach the image of the Developer Tool:
[image]

The configuration of the CamServer and CamViz its okey, its the same as your image.

PabloMorenoVera, 2018-05-24:
Sorry,

I attach the image of the terminals and the console of both programs (Camserver and CamViz):
[image]

Thanks!

Pablo

rperezgonz, 2018-05-24:
To try to detect where the problems may be, Can you do some tests?:

1. CamVizWeb + ros usb cam (https://jderobot.org/Tutorials#Ros_usb_cam_.2B_Cameraview-web).

2. CamVizWeb with nodejs and CamserverWeb with nodejs (you don't run **npm start**, run **node run.js** and then open it in a browser.

3. Try to run CamViz with ICE and camserver (https://jderobot.org/Tutorials#Camserver_.2B_CamViz and Change the instructions to run CamViz by CamViz Web).

With this, we will can know where is the problem, if it's in camviz or in camserver. Any questions you have you tell me

Thanks.

PabloMorenoVera, 2018-05-25:
Hi Roberto,
I comment the results of the test:
1. The test-launch instruction get mi image but cameraViz doesn't take the images.
2. In the second test, cameraserver connects to the rosbridge and set the camera but camViz doesn't connect to rosbridge.
3. In the third test, camserver and CamViz worked fine. CamViz get the images from the camera.

Thanks for all!

rperezgonz, 2018-05-28:
Ok, this is strange. It seem that The error is in CamViz and not in CamServer (the first test would have worked correctly).

Can you put in a new console "rostopic list" when CamServer and CamViz have already run? The only thing that I think may be failing is that the topic is not being sent. (This was my problem in the first version)

Thanks

PabloMorenoVera, 2018-05-29:
Hi Roberto!

When a run CamServer and CamViz and "Start" CamViz, the result of he "rostopic list" is the following:
[image]

Thanks!

rperezgonz, 2018-06-03:
Ok, we're going to try to do the last test. For this test, it's needed to install web_video_server.

1. sudo apt-get install ros-kinetic-web-video-server
2. roslaunch rosbridge_server rosbridge_websocket.launch
3. rosrun web_video_server web_video_server
4. Run CamServer (if the first doesn't run, try with the ros usb_cam) 
5. http://localhost:8080/stream?topic=/usb_cam/image_raw&type=ros_compressed

if it does not work with CamServer or with usb_cam, the problem is not related to CamServer or CamViz. Probably it will be of some problem in the ros installation. If this works well, it will check that it is failing

PabloMorenoVera, 2018-06-04:
Hi Rober,
I just run the last test and it works well, i show you an image:
[image]

Thanks!

Pablo,

closed 2021-02-21 by jmplaza