JdeRobot/base #1223: follow lines

fgonzalezr1998, 2018-06-15:
I've seen is not programmed the follow lines function at pibot.py for the real PiBot. Iḿ taking care of doing it, ok? I'm programming it similar than pibot.py for gazebo

jmplaza, 2018-06-15:
Hi @fgonzalezr1998 better that the follow-line function, it would be better to provide the functions to access the required sensors if they are not already included. Follow-line is a behavior and so it should not be included in the driver for PiBot, it is just an exercise on JdeRobot-Kids set. The driver only contains the functions to access to robot sensors and robot actuators.

jmplaza, 2018-06-16:
Ok @fgonzalezr1998, I got it now. The idea is to provide support for reading values of two IR sensors looking downwards in the real PiBot. They are supported in the simulated PiBot, but not yet in the real one. This support is quite convenient for the follow-line behavior. I rename the issue accordingly.

closed 2018-07-24 by fqez