JdeRobot/base #1306: [PiBot] - Caster redefinition

igarag, 2018-11-27:


igarag, 2018-11-27:
Make the wheel part printable.

fgonzalezr1998, 2018-12-02:
I propose two options:
1) We can to print a piece like [this](https://www.thingiverse.com/thing:2565379) and buy a caster like [this](https://www.amazon.es/Esfera-Zyurong-inoxidable-pulido-decoraci%C3%B3n/dp/B07CHF2PGX/ref=sr_1_6?ie=UTF8&qid=1543758277&sr=8-6&keywords=esfera%2Bmetal%2Brodamiento&th=1)

2) We can buy [this caster](https://tienda.bricogeek.com/robotica/108-rueda-loca-metalica-3-8.html?search_query=rueda+loca&results=3)

igarag, 2018-12-07:
Yeah, something like that is what we're looking for. We did a little test with a piece we designed that only served as support. We will consider this possibility. Thank you ;-)

closed 2021-02-21 by jmplaza