JdeRobot/base #1339: question follow_road

luiscaiza, 2019-02-11:
Hi everyone,

In order to launch gazebo-mavros-solo with two cameras, I did everything from https://github.com/JdeRobot/JdeRobot/blob/9551d91c3f04fd5ff909c5d9f971c2d8f3bac080/src/drivers/gazebo/plugins/solo/README.md , but launching It doesn't work, the message says:

luis@lc-desktop:/opt/jderobot/share/jderobot/gazebo/launch$ roslaunch follow_road.launch [follow_road.launch] is not a launch file name
The traceback for the exception was written to the log file

anyone can help me?

fqez, 2019-02-12:
Hi @luiscaiza it seems like you don't have the launch file installed in that directory. Can you share the content of your `/opt/jderobot/share/jderobot/gazebo/launch` directory?
If it's not installed you only have to copy (this file)[https://github.com/JdeRobot/JdeRobot/blob/9551d91c3f04fd5ff909c5d9f971c2d8f3bac080/assets/gazebo/launch/follow_road.launch]  to that directory and try again.
Let me know if you succeed!

luiscaiza, 2019-02-12:
Thanks Francisco,

I did what you say, but know, the systems can not find _solodrone_ file. The message from console is:

cp: no se puede efectuar `stat' sobre '/opt/jderobot/share/jderobot/drivers/solodrone': No existe el archivo o el directorio
process[px4-5]: started with pid [20393]
/opt/ros/kinetic/share/mavros/px4.sh: línea 12: cd: /tmp/solo: No existe el archivo o el directorio
/opt/ros/kinetic/share/mavros/px4.sh: línea 14: ./px4: No existe el archivo o el directorio
[px4-5] process has died [pid 20393, exit code 127, cmd /opt/ros/kinetic/share/mavros/px4.sh __name:=px4 __log:=/home/luis/.ros/log/c139489e-2ee6-11e9-ba18-588a5a1c700b/px4-5.log].
log file: /home/luis/.ros/log/c139489e-2ee6-11e9-ba18-588a5a1c700b/px4-5*.log

thank you again....

fqez, 2019-02-13:
hmm weird.
Can you please paste here the content of your following directories:
```
/opt/jderobot/share/jderobot/gazebo/plugins/solo
/opt/jderobot/share/jderobot/gazebo/models
```
And the content of your `/opt/jderobot/share/jderobot/gazebo/gazebo-setup.sh` file?
Also, make sure you have executed the following instruction, since it seems that driver is not finding px4.sh file:

```sudo cp ~/jderobot/src/drivers/gazebo/plugins/solo/px4.sh /opt/ros/kinetic/share/mavros```

luiscaiza, 2019-02-13:
here are the contents:

luis@lc-desktop:/opt/jderobot/share/jderobot/gazebo/plugins/solo$ ls
CommandMotorSpeed.pb.cc  Imu.pb.h          Odometry.pb.cc     README.md
CommandMotorSpeed.pb.h   Int32.pb.cc       Odometry.pb.h      setup.sh
Float.pb.cc              Int32.pb.h        OpticalFlow.pb.cc  SITLGps.pb.cc
Float.pb.h               IRLock.pb.cc      OpticalFlow.pb.h   SITLGps.pb.h
Groundtruth.pb.cc        IRLock.pb.h       px4.sh
Groundtruth.pb.h         MotorSpeed.pb.cc  Range.pb.cc
Imu.pb.cc                MotorSpeed.pb.h   Range.pb.h

luis@lc-desktop:/opt/jderobot/share/jderobot/gazebo/models$ ls
acera                   f1_tororosso_laser        monacoLine
amazon_warehouse_robot  f1_virgin                 mud_box
ArDrone2                f1_virgin_laser           newF1
arrow                   f1_williams               nurburgrinLine
blocks                  f1_williams_laser         opel
car                     face1                     opel_ROS
carApriltag             face2                     paredesMontmelo
carColorBeacon          face3                     pibot
car_holo                face4                     pibot_us
carNoMotor              face5                     pioneer2dx
cereales                face6                     pioneer2dxJde
circuitoMBOT            flyingKinect2             pista
circuitoMontmelo        formula1                  pistaSimple
cityLarge               gas_station               polaris_ranger_ev
cityMedium              grass_plane               RedArDrone2
construction_barrel     grass_plane2              rescue_area
construction_cone       GreenArDrone2             room
create                  ground_plane_sincolor     roomba
createROS               ground_plane_transparent  roombaROS
charactersMario         hangar                    simpleWorld
departamental           hokuyo                    solo
duck                    house_1                   speed_limit_sign
f1                      house_2                   stop_sign
f1_dummy                house_3                   stopW
f1_findia               house_4                   sun
f1_findia_laser         house_int2                taxi_holo
f1_hrt                  iris_fpv_cam              taxi_holo_Laser
f1_hrt_laser            jarama                    taxi_holo_laser_ROS
f1_lotus                kinect                    taxi_holo_ROS
f1_lotus_laser          laberinto                 tree
f1_mclaren              labyrinth                 turtlebotJde2cam
f1_mclaren_laser        lamp_post                 turtlebotROS
f1_mercedes             logoDeloitte              vrc_heightmap_1
f1_mercedes_laser       logoETSIT                 wall1
f1_redbull              logoGMV                   wall2
f1_redbull_laser        logoJdeRobot              wall3
f1_renault              logoURJC                  wall4
f1_renault_laser        mailbox                   warehouse_pallet
f1ROS                   mapaSimple                warehouse_walls
f1_sauber               monaco                    whiteTaxi
f1_sauber_laser         monacoElevations          yellowTaxi
f1_tororosso            monacoElevationsLine

# gazebo-setup fixed by varribas at 2016-04-08
# official Gazebo resources
. /usr/share/gazebo/setup.sh
# JdeRobot resources
export
GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:/opt/jderobot/share/jderobot/gazebo/plugins/solo
export
GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/share/jderobot/gazebo/plugins/solo
export
GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/opt/jderobot/share/jderobot/gazebo/plugins/flyingKinect:/opt/jderobot/share/jderobot/gazebo/plugins/dummycar:/opt/jderobot/share/jderobot/gazebo/plugins/car:/opt/jderobot/share/jderobot/gazebo/plugins/holo_car:/opt/jderobot/share/jderobot/gazebo/plugins/taxiholoLaser:/opt/jderobot/share/jderobot/gazebo/plugins/formula1:/opt/jderobot/share/jderobot/gazebo/plugins/pibot:/opt/jderobot/share/jderobot/gazebo/plugins/turtlebot:/opt/jderobot/share/jderobot/gazebo/plugins/roomba:/opt/jderobot/share/jderobot/gazebo/plugins/f1:/opt/jderobot/share/jderobot/gazebo/plugins/opel:/opt/jderobot/share/jderobot/gazebo/plugins/gazebo_ros_screw_plugin:/opt/jderobot/share/jderobot/gazebo/plugins/pioneer:/opt/jderobot/share/jderobot/gazebo/plugins/kinect:/opt/jderobot/share/jderobot/gazebo/plugins/gymkhana:/opt/jderobot/share/jderobot/gazebo/plugins/quadrotor
export
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/jderobot/share/jderobot/gazebo/plugins/flyingKinect:/opt/jderobot/share/jderobot/gazebo/plugins/dummycar:/opt/jderobot/share/jderobot/gazebo/plugins/car:/opt/jderobot/share/jderobot/gazebo/plugins/holo_car:/opt/jderobot/share/jderobot/gazebo/plugins/taxiholoLaser:/opt/jderobot/share/jderobot/gazebo/plugins/formula1:/opt/jderobot/share/jderobot/gazebo/plugins/pibot:/opt/jderobot/share/jderobot/gazebo/plugins/turtlebot:/opt/jderobot/share/jderobot/gazebo/plugins/roomba:/opt/jderobot/share/jderobot/gazebo/plugins/f1:/opt/jderobot/share/jderobot/gazebo/plugins/opel:/opt/jderobot/share/jderobot/gazebo/plugins/gazebo_ros_screw_plugin:/opt/jderobot/share/jderobot/gazebo/plugins/pioneer:/opt/jderobot/share/jderobot/gazebo/plugins/kinect:/opt/jderobot/share/jderobot/gazebo/plugins/gymkhana:/opt/jderobot/share/jderobot/gazebo/plugins/quadrotor

I have executed  sudo cp ~/jderobot/src/drivers/gazebo/plugins/solo/px4.sh
/opt/ros/kinetic/share/mavros.
Just inside px4.sh file there is a command line: cp -r
/opt/jderobot/share/jderobot/drivers/solodrone $solodir .
The solodrone file I do not have in my repository. Futhermore, the drivers
folder there is not.

thank you for helping me

fqez, 2019-02-13:
it seems like you are missing a step of the instructions. In your gazebo-setup.sh you should have `:/opt/jderobot/share/jderobot/gazebo/plugins/solo` at the end of the `GAZEBO_PLUGIN_PATH and LD_LIBRARY_PATH`
so in your gazebo-setup.sh file you should have the following:

GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/opt/jderobot/share/jderobot/gazebo/plugins/flyingKinect:/opt/jderobot/share/jderobot/gazebo/plugins/dummycar:/opt/jderobot/share/jderobot/gazebo/plugins/car:/opt/jderobot/share/jderobot/gazebo/plugins/holo_car:/opt/jderobot/share/jderobot/gazebo/plugins/taxiholoLaser:/opt/jderobot/share/jderobot/gazebo/plugins/formula1:/opt/jderobot/share/jderobot/gazebo/plugins/pibot:/opt/jderobot/share/jderobot/gazebo/plugins/turtlebot:/opt/jderobot/share/jderobot/gazebo/plugins/roomba:/opt/jderobot/share/jderobot/gazebo/plugins/f1:/opt/jderobot/share/jderobot/gazebo/plugins/opel:/opt/jderobot/share/jderobot/gazebo/plugins/gazebo_ros_screw_plugin:/opt/jderobot/share/jderobot/gazebo/plugins/pioneer:/opt/jderobot/share/jderobot/gazebo/plugins/kinect:/opt/jderobot/share/jderobot/gazebo/plugins/gymkhana:/opt/jderobot/share/jderobot/gazebo/plugins/quadrotor:**/opt/jderobot/share/jderobot/gazebo/plugins/solo**

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/jderobot/share/jderobot/gazebo/plugins/flyingKinect:/opt/jderobot/share/jderobot/gazebo/plugins/dummycar:/opt/jderobot/share/jderobot/gazebo/plugins/car:/opt/jderobot/share/jderobot/gazebo/plugins/holo_car:/opt/jderobot/share/jderobot/gazebo/plugins/taxiholoLaser:/opt/jderobot/share/jderobot/gazebo/plugins/formula1:/opt/jderobot/share/jderobot/gazebo/plugins/pibot:/opt/jderobot/share/jderobot/gazebo/plugins/turtlebot:/opt/jderobot/share/jderobot/gazebo/plugins/roomba:/opt/jderobot/share/jderobot/gazebo/plugins/f1:/opt/jderobot/share/jderobot/gazebo/plugins/opel:/opt/jderobot/share/jderobot/gazebo/plugins/gazebo_ros_screw_plugin:/opt/jderobot/share/jderobot/gazebo/plugins/pioneer:/opt/jderobot/share/jderobot/gazebo/plugins/kinect:/opt/jderobot/share/jderobot/gazebo/plugins/gymkhana:/opt/jderobot/share/jderobot/gazebo/plugins/quadrotor:**/opt/jderobot/share/jderobot/gazebo/plugins/solo**

Try to modify it and then do:

`source /opt/jderobot/share/jderobot/gazebo/gazebo-setup.sh` 

and launch it again.

luiscaiza, 2019-02-13:
Sorry,

But I am having the same problem. The message from console is:

process[master]: started with pid [23292]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 91e8e9a2-2fc0-11e9-9b22-588a5a1c700b
process[rosout-1]: started with pid [23305]
started core service [/rosout]
process[gazebo-2]: started with pid [23330]
process[gazebo_gui-3]: started with pid [23335]
process[mavros-4]: started with pid [23339]
process[px4-5]: started with pid [23341]

cp: no se puede efectuar `stat' sobre '/opt/jderobot/share/jderobot/drivers/solodrone': No existe el archivo o el directorio
/opt/ros/kinetic/share/mavros/px4.sh: línea 12: cd: /tmp/solo: No existe el archivo o el directorio
/opt/ros/kinetic/share/mavros/px4.sh: línea 14: ./px4: No existe el archivo o el directorio
[px4-5] process has died [pid 23341, exit code 127, cmd /opt/ros/kinetic/share/mavros/px4.sh __name:=px4 __log:=/home/luis/.ros/log/91e8e9a2-2fc0-11e9-9b22-588a5a1c700b/px4-5.log].
log file: /home/luis/.ros/log/91e8e9a2-2fc0-11e9-9b22-588a5a1c700b/px4-5*.log

fqez, 2019-02-13:
yeah i forgot to mention about that part of your problem, sorry. Try to modify the script px4.sh changing the solodrone to solo at the `cp` command line.

I hope this solves all your problems.

luiscaiza, 2019-02-14:
hi Francisco,

In the path:  /opt/jderobot/share/jderobot/drivers/solo
I do not have the solo file, but it is in the path
/opt/jderobot/share/jderobot/gazebo/plugins
do you recomend to change the path?

Another thing, in the px4.sh, there is a command line  ./px4 .
./posix-configs/SITL/init/ekf2/iris
In my directories there are not any folder like that. I think it is a old
version of Firmware, I just have the new version of Firmware of px4
can you tell how find it?

thanks

luiscaiza, 2019-02-25:
It is me again,

I can not solve my problem with the  follow_road exercise.
can you tell me what version of Firmware of px4 is used for the exercise? I have the newest version, in this,  there is not the file /posix-configs/SITL/init/ekf2/iris

thanks

nkhedekar, 2020-03-26:
Fixed in the migration project of GSoC 2019

closed 2020-03-26 by nkhedekar