JdeRobot/base #1341: Tile wheel for PiBot

jmvega, 2019-02-25:
We need to find a better solution for this issue.

jmvega, 2019-02-25:
I've been trying to find a tail wheel for the PiBot similar to the one I used at the beginning of this project, which was by Parallax and I found it in its website [2] but, as you can see, it is very expensive (4$ from US). So, I have also tried to find cheaper solutions... Aliexpress! And I found a couple of very interesting options [3] and [4].

  In short, I would order about 20 units of no. [3].

  Bests,

  Julio

[1] https://github.com/JdeRobot/base/issues/1341
[2] https://www.parallax.com/product/700-00009
[3] https://es.aliexpress.com/item/W420-de-acero-Omni-bola-Universal-de-omnidireccional-ruedas-para-Arduino-DIY-Robot-inteligente-coche-Eduational/32859925627.html
[4] https://es.aliexpress.com/item/Metal-Caster-Universal-Wheel-Eye-Ball/32674846005.html

igarag, 2019-02-25:
Hi Julio.

Indeed, the second option is very interesting. We already had the third option in the first prototype and it's very heavy, expensive and big. I think it's a good idea to add the [second option](https://es.aliexpress.com/item/W420-de-acero-Omni-bola-Universal-de-omnidireccional-ruedas-para-Arduino-DIY-Robot-inteligente-coche-Eduational/32859925627.html) to the list of components.

Regards.
Nacho.

closed 2021-02-21 by jmplaza