JdeRobot/base #1342: [question] Drone infraestructure installation problem

rodrigo-pacheco, 2019-02-25:
Hi all,

I am trying to install new drone infraestructure that makes use of MavLink -PX4-, MavROS and Gazebo. I have been following the steps mentioned [here](https://jderobot.org/Drones) but could not make it to Stage 2. 

[Here are the commands](https://github.com/RoboticsURJC-students/2018-tfg-rodrigo-pacheco/blob/master/ENVIRONMENT_INTALLATION/intallation_receipt.txt) I have executed successfullly to try and install the infraestructure: 

When trying to arm 3DR Solo motors with [ rosrun mavros mavsafety arm ] I get the following message: 
rodrigo:~$ rosrun mavros mavsafety arm
Request failed. Check mavros logs
rodrigo:~$ rosrun mavros mavcmd takeoff 5 5 5 5
usage: mavcmd takeoff [-h] min_pitch yaw latitude longitude altitude
mavcmd takeoff: error: the following arguments are required: altitude
rodrigo:~$ rosrun mavros mavcmd takeoff 5 5 5 5 5
Request failed. Check mavros logs. ACK: 4

It seems like connection is not carried out properly, even though link between PX4 SITL and MavROS node obtained correctly [ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557" ]:

[image]

[This is the output ](https://github.com/RoboticsURJC-students/2018-tfg-rodrigo-pacheco/blob/master/ENVIRONMENT_INTALLATION/mavros_logs_stage_1.txt) of executing [ rostopic echo -n1 /diagnostics | tee /tmp/diag.yml ]. I have tried finding some help but links [like this one](https://github.com/mavlink/mavros/issues/538) or [this other one](https://github.com/mavlink/mavros/issues/719) haven't been very helpful.

Thank you in advanced! :smile:

rodrigo-pacheco, 2019-03-04:
Hello again,

I think that the problem I could be facing is that I don't launch Gazebo world correctly. Therfore, connection is not carried out successfully and engines aren't armed.
Could you tell me where to find the world with the quadricopter shown in the video and how it has been launched in Gazebo?

Thank you very much!

nkhedekar, 2020-03-26:
Fixed in the Migration GSoC 2019 project

closed 2020-03-26 by nkhedekar