JdeRobot/base #1343: Support for Ubuntu 18.04 LTS

vinay0410, 2019-03-04:
Support for both Source build and packages for Ubuntu 18.04 LTS

sleep-404, 2019-03-07:
would love to work on this !
But while checking for dependencies using cmake , I am having little trouble with Nite and Nite2 deps , from where can I install them ? 
Would be of great help if you can point out a few resources !!

vinay0410, 2019-03-11:
Hi @sleep-404 ,
As far as I know those dependencies are optional, so I think you can build JdeRobot without them too.
And the first step would be to provide support to source build on Ubuntu 18.04.

Then you can move to building debian packages

fqez, 2019-03-11:
Hi guys,

there's work done on this matter already. It's just testing about gazebo plugins refactorization, ICE compilation fixings, etc. But I think it's a good starting point.

The last two commits on [this branch](https://github.com/JdeRobot/base/tree/ubuntu_bionic) has them. Sorry about the mess in the commits and the lack of documentation, it's just a proof of concept. I will update the new third party dependencies in this same thread, so you can have a valid environmet to start migrating the project.

If you want to start from scratch tho, I suggest you to review the changes in ZeroC ICE library (from v3.6 to v3.7) because the **IceUtil** module doesn't exist anymore and it's one of the "pillars" of our compilation chain. This ICE issue is "solved" in that branch I linked you in the previous paragraph tho.

Thanks for your help!

sleep-404, 2019-03-17:
@fqez  , while building I'm getting this error
Please set them or make sure they are set and tested correctly in the CMake files:
OpenCV_INCLUDE_DIR
[image]
It was working just fine till today , 
but as my package manager got crashed I re-installed ubuntu 16.04 , and following the the similar steps as before , gives me the above error.
I have already installed packages related to opencv , by searching them using 
`apt search opencv`
and did find where they are using 
`pkg-config --libs opencv`
but don't which one to use as the above command gave a lot of places

fqez, 2019-03-21:
Hi, @sleep-404 
do you mean you reinstalled 16.04 or 18.04? There is a "huge" difference in JdeRobot installation steps between versions regarding to OpenCV. In Ubuntu 16.04, we do not install official OpenCV packages but ROS ones to compile JdeRobot due to some incompatibilities we found back in the day. On the other hand, for Ubuntu 18.04, we do install official OpenCV packages, since it is compatible with our compilation chain.
If you are referring to ubuntu 16, please move your question to another issue. If it's a typo, it should work with official opencv packages. I will try to post here the packages I used for compiling jderobot in Ubuntu 18, which worked for me.

Regards!

sleep-404, 2019-03-21:
@fqez , installed it successfully . But I'm pretty sure I did not post it under this issue , don't know how it came here. Any how everything works fine now and thank you for the your detailed explanation and time !

closed 2019-05-23 by fqez