JdeRobot/base #1344: Error while loading shared libraried:libeasyiceconfig

SamPusegaonkar, 2019-03-09:
I am trying to run a camserver on my laptop using this command.

$ camserver camserver.cfg

But I got this error:
camserver: error while loading shared libraries: libeasyiceconfig.so: cannot open shared object file: No such file or directory

jiban-jyoti, 2019-03-09:
Hello Sam, It seems like you have not installed the ICE dependency for Jderobot. So, kindly install all the dependencies.

abhiMishra98, 2019-03-09:
Install dependencies:
sudo apt install libdb5.3-dev libdb5.3++-dev libssl-dev libbz2-dev libmcpp-dev
compile ice:
git clone -b 3.6 https://github.com/zeroc-ice/ice.git 
cd ice/cpp
make CPP11=yes OPTIMIZE=yes
make install

SamPusegaonkar, 2019-03-09:
Hey @abhiMishra98, thanks for your response...
After I run the make CPP11-yes OPITMIZE=yes command I get this (see below ). I fail to understand where the error lies.

sameer@sameer-laptop:~/Desktop/ice/cpp$ make CPP11=yes OPTIMIZE=yes
making all in config
make[1]: Entering directory '/home/sameer/Desktop/ice/cpp/config'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/sameer/Desktop/ice/cpp/config'
making all in src
make[1]: Entering directory '/home/sameer/Desktop/ice/cpp/src'
making all in IceUtil
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/IceUtil'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/IceUtil'
making all in Slice
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/Slice'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/Slice'
making all in slice2cpp
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2cpp'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2cpp'
making all in slice2cs
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2cs'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2cs'
making all in slice2freeze
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2freeze'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2freeze'
making all in slice2freezej
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2freezej'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2freezej'
making all in slice2java
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2java'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2java'
making all in slice2php
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2php'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2php'
making all in slice2py
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2py'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2py'
making all in slice2html
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2html'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2html'
making all in slice2js
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/slice2js'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/slice2js'
making all in Ice
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/Ice'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/Ice'
making all in IceXML
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/IceXML'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/IceXML'
making all in IceSSL
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/IceSSL'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/IceSSL'
making all in IceDiscovery
make[2]: Entering directory '/home/sameer/Desktop/ice/cpp/src/IceDiscovery'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/src/IceDiscovery'
[... 298 lines cut ...]
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/sameer/Desktop/ice/cpp/test/Ice/threadPoolPriority'
making all in invoke
make[3]: Entering directory '/home/sameer/Desktop/ice/cpp/test/Ice/invoke'
rm -f client
g++ -rdynamic -std=c++0x -m64 -fvisibility=hidden -Wall -Werror -pthread -fPIC -O2 -DNDEBUG -L../../../lib/x86_64-linux-gnu/c++11 -Wl,--disable-new-dtags -Wl,-rpath,\$ORIGIN/../../../lib/x86_64-linux-gnu -o client Test.o Client.o AllTests.o  -lIce -lIceUtil
AllTests.o: In function `allTests(IceInternal::Handle<Ice::Communicator> const&)':
AllTests.cpp:(.text+0x5064): undefined reference to `IceProxy::Ice::Object::__begin_ice_invoke(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ice::OperationMode, std::vector<unsigned char, std::allocator<unsigned char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const*, IceInternal::Function<void (bool, std::vector<unsigned char, std::allocator<unsigned char> > const&)> const&, IceInternal::Function<void (IceUtil::Exception const&)> const&, IceInternal::Function<void (bool)> const&)'
AllTests.cpp:(.text+0x54d4): undefined reference to `IceProxy::Ice::Object::__begin_ice_invoke(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ice::OperationMode, std::pair<unsigned char const*, unsigned char const*> const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::alloca [... cut ...]
Makefile:32: recipe for target 'client' failed
make[3]: *** [client] Error 1
make[3]: Leaving directory '/home/sameer/Desktop/ice/cpp/test/Ice/invoke'
Makefile:65: recipe for target 'invoke' failed
make[2]: *** [invoke] Error 2
make[2]: Leaving directory '/home/sameer/Desktop/ice/cpp/test/Ice'
Makefile:35: recipe for target 'Ice' failed
make[1]: *** [Ice] Error 2
make[1]: Leaving directory '/home/sameer/Desktop/ice/cpp/test'
Makefile:51: recipe for target 'all' failed
make: *** [all] Error 1

Thanks

SamPusegaonkar, 2019-03-09:
@jiban-jyoti Thanks for your short explanation!

vinay0410, 2019-03-11:
Hi @SamPusegaonkar ,
Is this resolved.
If not add zeroc ice resources as mentioned in this link: https://jderobot.org/Installation#Install_the_dependences

and follow the instructions carefully, you've surely missed something.

Also, we don't support builds for Ubuntu 18.04 yet, so if you were installing on 18.04, them that micght be the reason for the above error

fqez, 2019-03-12:
HI @SamPusegaonkar 
are you sure that dynamic library is installed in your system? If you installed JdeRobot in the default path, check `/opt/jderobot/lib`and make sure `libeasyiceconfig.so` is there.
If it's there, make sure you sourced the jderobot setup:
```
$ source /opt/jderobot/setup.bash
```
and try again. Hope it helps!

SamPusegaonkar, 2019-03-12:
@fqez The 'libeasyiceconfig.so' is present in the path you mentioned. I also sourced the jderobot setup.
I'm getting the same error.

EL-SHREIF, 2019-03-14:
Hello I have this error when run this command 
make CPP11=yes OPTIMIZE=yes
============================================================================
it gives me :
el-shreif@elshreif-PC:~/jde/base/ice/cpp$ make CPP11=yes OPTIMIZE=yes
making all in config
make[1]: Entering directory '/home/el-shreif/jde/base/ice/cpp/config'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/config'
making all in src
make[1]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src'
making all in IceUtil
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/IceUtil'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/IceUtil'
making all in Slice
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/Slice'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/Slice'
making all in slice2cpp
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2cpp'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2cpp'
making all in slice2cs
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2cs'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2cs'
making all in slice2freeze
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2freeze'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2freeze'
making all in slice2freezej
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2freezej'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2freezej'
making all in slice2java
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2java'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2java'
making all in slice2php
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2php'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2php'
making all in slice2py
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2py'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2py'
making all in slice2html
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2html'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2html'
making all in slice2js
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/slice2js'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/slice2js'
making all in Ice
make[2]: Entering directory '/home/el-shreif/jde/base/ice/cpp/src/Ice'
g++ -c -I.. -I../../include -DICE_API_EXPORTS  -std=c++0x -m64 -fvisibility=hidden -Wall -Werror -pthread -fPIC -O2 -DNDEBUG -MMD NetworkProxy.cpp -MF .depend/NetworkProxy.d
NetworkProxy.cpp: In constructor ‘{anonymous}::SOCKSNetworkProxy::SOCKSNetworkProxy(const string&, int)’:
NetworkProxy.cpp:79:42: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘union IceInternal::Address’; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&_address, 0, sizeof(_address));
                                          ^
In file included from ../Ice/NetworkProxy.h:13,
                 from NetworkProxy.cpp:10:
../Ice/Network.h:125:7: note: ‘union IceInternal::Address’ declared here
 union Address
       ^~~~~~~
NetworkProxy.cpp: In constructor ‘{anonymous}::HTTPNetworkProxy::HTTPNetworkProxy(const string&, int)’:
NetworkProxy.cpp:192:42: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘union IceInternal::Address’; use assignment or value-initialization instead [-Werror=class-memaccess]
     memset(&_address, 0, sizeof(_address));
                                          ^
In file included from ../Ice/NetworkProxy.h:13,
                 from NetworkProxy.cpp:10:
../Ice/Network.h:125:7: note: ‘union IceInternal::Address’ declared here
 union Address
       ^~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [../../config/Make.rules:386: NetworkProxy.o] Error 1
make[2]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src/Ice'
make[1]: *** [Makefile:91: Ice] Error 2
make[1]: Leaving directory '/home/el-shreif/jde/base/ice/cpp/src'
make: *** [Makefile:51: all] Error 1
  
===============================================================================
how to solve this ??

sleep-404, 2019-03-18:
Hi @SamPusegaonkar , 
can you please share the following information 
1. your `cmake ..` output ?
2. which repository are you cloning and 
3. Your operating system & version

SamPusegaonkar, 2019-03-21:
Done compiling with a fresh install of Ubuntu 16.04! :) Thanks everyone

closed 2019-03-21 by SamPusegaonkar