JdeRobot/base #1345: JdeRobot installation : missing packages

Mintu98, 2019-03-09:
While installing Jderobot, i could not download packages for NITE2,NITE,libfreenect and PCL_apps.
Please help me with the links to download the missing packages.

sleep-404, 2019-03-11:
yeah even I'm facing similar problem

lr-morales, 2019-03-11:
Assuming an Ubuntu system, for libfreenect and pcl, please check first that "universe" repository is enabled and updated. Then search for libfreenect-dev and libpcl-dev packages, which should contain those requirements.

As for NITE and NITE2, can't check for instructions at this moment, sorry.

vinay0410, 2019-03-11:
Hi @Mintu98 ,
Could you post the end of this CMake output. Nite Dependencies are probably optional and I believe something else is causing this error

sleep-404, 2019-03-18:
Hi @Mintu98 ,
Can you now build the repository ?

Mintu98, 2019-03-18:
Thanks @vinay0410 it helped and the problem is resolved. Thanks again! :)

closed 2019-03-18 by Mintu98