JdeRobot/base #1347: error at executing make after installing all dependencies

teamleader6, 2019-03-11:
[ 45%] Linking CXX executable recorder2
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFWriteEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFNumberOfStrips@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFClose@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFOpen@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
src/tools/recorder2/CMakeFiles/recorder2.dir/build.make:595: recipe for target 'src/tools/recorder2/recorder2' failed
make[2]: *** [src/tools/recorder2/recorder2] Error 1
CMakeFiles/Makefile2:4092: recipe for target 'src/tools/recorder2/CMakeFiles/recorder2.dir/all' failed
make[1]: *** [src/tools/recorder2/CMakeFiles/recorder2.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Aman-py, 2019-03-12:
this jderobot official installation guide go through this and try this

[jdeRobot Install Documentation](https://jderobot.org/Installation)

or try to install according to this video tutorial

[jdeRobot Installation](https://www.youtube.com/watch?v=R-0AEXS3Wj0)

teamleader6, 2019-03-12:
i did like in documentation and video also

fqez, 2019-03-12:
Hi @teamleader6 

did you source the ROS setup before compiling the project? This is required for the linking process since the OpenCV build we are currently using is the ROS one.

Assuming you have all the dependencies installed, the steps should be the following:

1. Source ROS setup: 
```
$ source /opt/ros/kinetic/setup.bash
```
2. Complie the project (out of source build recommended)
```
$ cd ~/base && mkdir build && cd build
$ cmake .. && make
```
3. Install it
```
$ sudo make install
```

Hope it helps!

teamleader6, 2019-03-12:
same error unfortunatly

fqez, 2019-03-12:
Can you summarize your environment here? Like OS you are working on, version of ROS installed, etc.
Make sure you deleted all the files generated on your previous build process before building again. CMakeCache could be betraying you. I just compiled the project in my machine Ubuntu-16.04 without issues.
Nonetheless, you have debian [packages available](https://jderobot.org/Installation#From_Debian_packages) for jderobot, there's no need to compile unless you are trying to install it in another OS.

teamleader6, 2019-03-12:
ubuntu 16.04 64bit rosversion kinetic python 3.7

closed 2019-03-13 by teamleader6

nguyen37, 2019-07-25:
@teamleader6 Hi! How were you able to get past this problem?

fqez, 2019-07-26:
Hi @nguyen37, 

what exact problem are you facing? Could you describe it so I reopen this issue if it fits or create a new one?

Thanks!