JdeRobot/base #1348: stuck at executing make after installing all dependencies as installation in documentation i compiled cameraserver and cameraviewer with no problem but stuck at compiling whole project

teamleader6, 2019-03-13:
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFWriteEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFNumberOfStrips@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFClose@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFOpen@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
/opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
src/tools/recorder2/CMakeFiles/recorder2.dir/build.make:595: recipe for target 'src/tools/recorder2/recorder2' failed
make[2]: *** [src/tools/recorder2/recorder2] Error 1
CMakeFiles/Makefile2:4092: recipe for target 'src/tools/recorder2/CMakeFiles/recorder2.dir/all' failed
make[1]: *** [src/tools/recorder2/CMakeFiles/recorder2.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

sleep-404, 2019-03-14:
what is your output of `cmake ..` , can you share it ?
Are all the configurations done ?

teamleader6, 2019-03-14:
it didn't give me configuration done because of conflict of anaconda3 and i solved it by run this command
conda deactivate and redoing compilation process
last lines of cmake output
deps-dev
dpkg-deb: building package 'jderobot-deps-dev' in '.dpkg_pkgs/jderobot-deps-dev_5.6.6_all.deb'.
drivers
dpkg-deb: building package 'jderobot-drivers' in '.dpkg_pkgs/jderobot-drivers_5.6.6_all.deb'.
examples
dpkg-deb: building package 'jderobot-examples' in '.dpkg_pkgs/jderobot-examples_5.6.6_all.deb'.
libs
dpkg-deb: building package 'jderobot-libs' in '.dpkg_pkgs/jderobot-libs_5.6.6_all.deb'.
tools
dpkg-deb: building package 'jderobot-tools' in '.dpkg_pkgs/jderobot-tools_5.6.6_all.deb'.
zeroc-ice-python
dpkg-deb: building package 'jderobot-zeroc-ice-python' in '.dpkg_pkgs/jderobot-zeroc-ice-python_5.6.6_all.deb'.
jderobot
dpkg-deb: building package 'jderobot' in '.dpkg_pkgs/jderobot_5.6.6_all.deb'.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mohamed/Desktop/base/build

closed 2019-03-14 by teamleader6

zmll, 2019-05-16:
i have same problem, can you tell  me the solution?