JdeRobot/base #1349: Compilation Error

EL-SHREIF, 2019-03-14:
When I run This command :
cmake .
=========================================================================
it gives me this error :
version:8.2.0
/home/el-shreif/jde/base/src/libs/jderobotutil
/home/el-shreif/jde/base/src/libs/colorspaces
/home/el-shreif/jde/base/src/libs/colorspacesmm
/home/el-shreif/jde/base/src/libs/easyiceconfig
/home/el-shreif/jde/base/src/libs/parallelIce
/home/el-shreif/jde/base/src/libs/depthLib
/home/el-shreif/jde/base/src/libs/geometry
/home/el-shreif/jde/base/src/libs/logger
/home/el-shreif/jde/base/src/libs/ns
/home/el-shreif/jde/base/src/libs/progeo
/home/el-shreif/jde/base/src/libs/pioneer
/home/el-shreif/jde/base/src/libs/fuzzylib
*** ALUT not found
-- Checking for one of the modules 'OPTIONAL;ardronelib'
-- ardronelib NOT FOUND.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
   -- CUDA not found!!
*** Fireware libraries: dc1394 or raw1394 not found
*** GAZEBO LIBRARIES FOUND: /usr/include/gazebo-9;/usr/include/sdformat-6.0;/usr/include/ignition/transport4;/usr/include/ignition/msgs1;/usr/include/ignition/fuel_tools1;/usr/include/x86_64-linux-gnu;/usr/lib/x86_64-linux-gnu/libzip/include;/usr/include/ignition/common1;/usr/include/ignition/math4;/usr/include/uuid;/usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include;/usr/include/x86_64-linux-gnu - 
-- Found glog    (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libglog.so)
*** Goocanvasmm LIBRARIES FOUND
CMake Error at Deps/ice/CMakeLists.txt:37 (MESSAGE):
  *** IceUtil not found
Call Stack (most recent call first):
  Deps/LoadJderobotDependencies.cmake:12 (include)
  CMakeLists.txt:162 (include)

-- Configuring incomplete, errors occurred!
See also "/home/el-shreif/jde/base/CMakeFiles/CMakeOutput.log".
See also "/home/el-shreif/jde/base/CMakeFiles/CMakeError.log".
=========================================================================
I tried to install zero-Ice lib with many ways but still have the same error

sleep-404, 2019-03-14:
you seem to be missing ice-utils libraries ,which can be installed using 
`sudo apt install jderobot-zeroc-ice-python
`
Hope that might get you going :)

EL-SHREIF, 2019-03-16:
it gives me :
>el-shreif@elshreif-PC:~/jde/base$  sudo apt install jderobot-zeroc-ice-python 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package jderobot-zeroc-ice-python
-----------------------------------------------------------------------------------------
but I install this :
>el-shreif@elshreif-PC:~/jde/base$ sudo apt install python-zeroc-ice
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-zeroc-ice is already the newest version (3.7.2-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
-------------------------------------------------------------------------------------------------
also when I try this to install all I got this :
>el-shreif@elshreif-PC:~/jde/base$ sudo apt install jderobot-deps-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package jderobot-deps-dev
-----------------------------------------------------------------------------
and I try to make sure that I have these:
[image]

jiban-jyoti, 2019-03-18:
Is this issue resolved ?

sleep-404, 2019-03-18:
@EL-SHREIF  ,
try adding Jderobot to the source list, using 

`sudo sh -c 'cat<<EOF>/etc/apt/sources.list.d/jderobot.list`
> `deb [arch=amd64] http://jderobot.org/apt xenial main` 
> `EOF'`

But as far as the docs are concerned , It is for 16.04, but I think you are using 18.04, even then try adding it . Might work.
And also add the public key from the JdeRobot repository  using
`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 24E521A4`
Comment back if you get any further errors

Pravin854, 2019-03-22:
`sudo apt install libzeroc-ice3.5-java`, this command worked for me

closed 2019-03-23 by EL-SHREIF