JdeRobot/base #1350: Missing packages during installation

gonidelis, 2019-03-14:
I have cloned the repo and tried to run
`cmake .` but I get this error:

> version:5.4.0
/home/john/JdeRobot/src/libs/jderobotutil
/home/john/JdeRobot/src/libs/colorspaces
/home/john/JdeRobot/src/libs/colorspacesmm
/home/john/JdeRobot/src/libs/easyiceconfig
/home/john/JdeRobot/src/libs/parallelIce
/home/john/JdeRobot/src/libs/depthLib
/home/john/JdeRobot/src/libs/geometry
/home/john/JdeRobot/src/libs/logger
/home/john/JdeRobot/src/libs/ns
/home/john/JdeRobot/src/libs/progeo
/home/john/JdeRobot/src/libs/pioneer
/home/john/JdeRobot/src/libs/fuzzylib
*** ALUT not found
-- Checking for one of the modules 'OPTIONAL;ardronelib'
-- ardronelib NOT FOUND.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
   -- CUDA not found!!
*** Fireware libraries: dc1394 or raw1394 not found
*** GAZEBO LIBRARIES FOUND: /usr/include/gazebo-7;/usr/include/sdformat-4.0;/usr/include/ignition/math2 - 
-- Found glog    (include: /usr/include, library: /usr/lib/x86_64-linux-gnu/libglog.so)
-- Checking for module 'gsl'
--   No package 'gsl' found
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  Deps/gsl/CMakeLists.txt:1 (PKG_CHECK_MODULES)
  Deps/LoadJderobotDependencies.cmake:12 (include)
  CMakeLists.txt:162 (include)

  

> --Configuring incomplete, errors occurred!
  See also "/home/john/JdeRobot/CMakeFiles/CMakeOutput.log".
  See also "/home/john/JdeRobot/CMakeFiles/CMakeError.log".

Any help on retrieving the missing packages would be appreciated.
Thanks in advance

sleep-404, 2019-03-15:
You are missing the gsl dependencies which is required for configuration.
Try installing it 
`sudo apt install gsl-bin` or `sudo apt install libgsl-dev`
One of them should work in your case. Comment back if you get any new errors.
And try installing all the JDE dependencies using
`sudo apt install jderobot-deps-dev`

gonidelis, 2019-03-15:
>Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

> The following packages have unmet dependencies:
 jderobot-deps-dev : Depends: zeroc-ice-slice but it is not installable
                                 Depends: libzeroc-ice3.6 but it is not installable
                                 Depends: zeroc-ice-utils but it is not installable
                                 Depends: libzeroc-icestorm3.6 but it is not installable
E: Unable to correct problems, you have held broken packages.

This is what I get when trying to run:
` sudo apt install jderobot-deps-dev`

gonidelis, 2019-03-15:
Even after running 
`sudo apt install jderobot-zeroc-ice-python`
When running `sudo apt install jderobot-deps-dev` I get this log:

> Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

> The following packages have unmet dependencies:
 jderobot-deps-dev : Depends: zeroc-ice-slice but it is not installable
                     Depends: libzeroc-ice3.6 but it is not installable
                     Depends: zeroc-ice-utils but it is not installable
                     Depends: libzeroc-icestorm3.6 but it is not installable
E: Unable to correct problems, you have held broken packages.

and I just can't figure out how to solve it.

jiban-jyoti, 2019-03-18:
Is this issue resolved or not?

gonidelis, 2019-03-18:
Unfortunately as I did not receive any answer I aborted source installation and chose to do binary installation which of course worked out.

sleep-404, 2019-03-18:
@gonidelis  , I think you might have exited the installation process in the middle that is the reason for that error.
Try following these tutorials:
[Unable to correct problems, you have held broken packages
](https://askubuntu.com/questions/363200/e-unable-to-correct-problems-you-have-held-broken-packages)[How to fix held broken packages?](https://unix.stackexchange.com/questions/389156/how-to-fix-held-broken-packages)

sleep-404, 2019-03-18:
@gonidelis  , which repository are you cloning.
Can you paste the command you used for cloning or the path where the build is ?

gonidelis, 2019-03-18:
The command was : `git clone https://github.com/JdeRobot/base.git`. Could that be the problem?

sleep-404, 2019-03-18:
Nope , it is perfectly fine

gonidelis, 2019-03-18:
As I told you I aborted source installation and went with:
`sudo apt install jderobot
sudo apt install jderobot-gazebo-assets`. I could try source installation again if you want to see if the problem is fixed, as long as this won't mess up the existed one.

sleep-404, 2019-03-18:
If your requirements are satisfied with instalment you need not do anything else. 
And if you plan to build from source read further :)
As long as you are using a package manager to do things nothing will mess up , atleast not until you close it abruptly. And **don't** use **sudo** unless your are working with `apt`.
So I guess you can do the following steps safely without messing anything:
 1.) `git clone https://github.com/JdeRobot/base.git`
 2.) `sudo apt install jderobot-deps-dev`
     This second step might give you errors(mentioned by you) , follow the suggestions given in the above links.That should fix them, else try googling the error ,you will definitely find many solutions
 3.)`sudo apt install jderobot-zeroc-ice-python`
 4.) `cd JdeRobot`
 5.) `mkdir build && cd build`
 6.) `cmake ..`
 7.) `make`
 8.) `make install`

gonidelis, 2019-03-18:
Thank you very much

gonidelis, 2019-03-18:
@sleep-404  I don't understand. I tried `git clone https://github.com/JdeRobot/base.git` and the folder that was created was named **base**, not **JdeRobot**. What's the difference? (I remember when I first source installed it, the folder that was created was name JdeRobot, I am pretty confused)

gonidelis, 2019-03-18:
Hello there, @sleep-404  
Trying to run `make` inside `build` folder got me this error:

> [ 65%] Building CXX object src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o
In file included from /home/john/base/src/drivers/picamserver/src/piCamCore.cpp:1:0:
/home/john/base/src/drivers/picamserver/src/piCamCore.h:3:33: fatal error: interface/mmal/mmal.h: No such file or directory
compilation terminated.
src/drivers/picamserver/CMakeFiles/picamserver.dir/build.make:110: recipe for target 'src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o' failed
make[2]: *** [src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o] Error 1
CMakeFiles/Makefile2:5056: recipe for target 'src/drivers/picamserver/CMakeFiles/picamserver.dir/all' failed
make[1]: *** [src/drivers/picamserver/CMakeFiles/picamserver.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

(ps: Sorry if I overdo it with this thread, should I open a new issue for this matter? I am just posting here as an answer to your previous post. Thank in advance)

sleep-404, 2019-03-19:
@gonidelis , don't worry , you have downloaded the correct repository only.
And before starting to build, your `cmake ..` should return configuration one.The output will look something like this [output_file.txt](https://github.com/JdeRobot/base/files/2981391/output_file.txt). See the end of the text file.
If it returns any errors , you should install the missing packages accordingly.
First check the CmakeLists.txt file look for the line
`FIND_PACKAGE(package-name)`
 and try searching for the package using
`apt search package-name`, if relevant results are found great, else you need to google the missing package to download them.
I did not downloaded any extra packages once I did `sudo apt install jderobot-deps-dev` apart from `zero-ice`.
Hope this helps.

gonidelis, 2019-03-19:
But as you can see the problem is in this line:

> In file included from /home/john/base/src/drivers/picamserver/src/piCamCore.cpp:1:0:
> /home/john/base/src/drivers/picamserver/src/piCamCore.h:3:33: fatal error: **interface/mmal/mmal.h: No such file or directory**
It's not a package that's missing, but a header file. 

That's the cmake file on this directory:

> src/drivers/picamserver/CMakeFiles/

[CMakeLists.txt](https://github.com/JdeRobot/base/files/2983210/CMakeLists.txt)

No ` FIND_PACKAGE(package-name)` line.

anidh1997, 2019-03-25:
I tried to solve the JdeRobot installation challenge but unable to install this base project. I installed all the missing libraries and modules but some unable to solve the errors found in image below. Someone please help. Thanks in advance.

[image]

PaulNicolasHunter, 2019-03-28:
@gonidelis is everything working for you?

I tried your steps to install :
```
sudo apt install jderobot
sudo apt install jderobot-gazebo-assets
```
but no luck, it gives me,

> 
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> E: Unable to locate package jderobot

and 

> 
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> E: Unable to locate package jderobot-gazebo-assets

any guesses why that is happening?
regards

gonidelis, 2019-03-30:
Hi @PaulNicolasHunter the problem is that you haven't added the necessary repositories. Check out in this page [( https://jderobot.org/Installation#From_Debian_packages)] the first part: **Add the repositories** before executing the `sudo apt install *` commands

PaulNicolasHunter, 2019-03-30:
@gonidelis thank you very much for your response, though i have another doubt that, while installing i saw that CUDA was also missing, does this also means that the program only works for Nvidia GPUs ?

Because I'm running UBUNTU on VM virtual box as guided by my mentor, and I have AMD on my host machine, even though I have enabled 3D acceleration, I'm skeptic that if it will run or not.
so, as everything is running well for you can you please tell me which GPU you have? and if it works on AMD?
Is this hardware dependent?

Pravin854, 2019-03-30:
Hi @anidh1997  try : 
'sudo apt-get install ardronelib'
'sudo apt-get install libgtkgl2.0-dev'
'sudo apt-get install libdc1394-22'

and @PaulNicolasHunter can you specify your system specification and steps which you have used to build it, because it is working for me.

PaulNicolasHunter, 2019-03-30:
@Pravin854 ,

Ubuntu 16.04 LTS , 64bit on VM VirtualBox.

Pravin854, 2019-03-30:
@PaulNicolasHunter , 
`wget https://zeroc.com/download/GPG-KEY-zeroc-release
sudo apt-key add GPG-KEY-zeroc-release
cd /etc/apt/sources.list.d
sudo wget https://zeroc.com/download/Ice/3.5/ubuntu/ice3.5-trusty.list
sudo apt-get update
sudo apt-get install ice icebox/trusty-zeroc icegrid/trusty-zeroc

sudo sh -c 'cat<<EOF>/etc/apt/sources.list.d/jderobot.list
deb [arch=amd64] http://jderobot.org/apt xenial main
EOF'
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 24E521A4
sudo apt update
sudo apt install jderobot
sudo apt install jderobot-gazebo-assets
source ~/.bashrc`

or can you specify what steps you have followed while build it from source and error which are getting while building it ?

Kareem-Emad, 2019-03-31:
@gonidelis  For the build error you mentioned 

```
[ 65%] Building CXX object src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o
In file included from /home/john/base/src/drivers/picamserver/src/piCamCore.cpp:1:0:
/home/john/base/src/drivers/picamserver/src/piCamCore.h:3:33: fatal error: interface/mmal/mmal.h: No such file or directory
compilation terminated.
src/drivers/picamserver/CMakeFiles/picamserver.dir/build.make:110: recipe for target 'src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o' failed
make[2]: *** [src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o] Error 1
CMakeFiles/Makefile2:5056: recipe for target 'src/drivers/picamserver/CMakeFiles/picamserver.dir/all' failed
make[1]: *** [src/drivers/picamserver/CMakeFiles/picamserver.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

Remove picamserver folder as it should not be in the build process.

anidh1997, 2019-03-31:
> Hi @anidh1997 try :
> `sudo apt-get install ardronelib`
> `sudo apt-get install libgtkgl2.0-dev`
> `sudo apt-get install libdc1394-22`

Thanks for your help, sir. I installed these above libraries but still, some major packages are missing which are obstructing in the installation. Missing libraries are in the picture. Please help, sir.

Thanks in advance.

[image]

Pravin854, 2019-03-31:
@anidh1997  Try,
`sudo apt-get install libraw1394-dev`

`wget https://zeroc.com/download/GPG-KEY-zeroc-release`
`sudo apt-key add GPG-KEY-zeroc-release`
`cd /etc/apt/sources.list.d`
`sudo wget https://zeroc.com/download/Ice/3.5/ubuntu/ice3.5-trusty.list`
`sudo apt-get update`
`sudo apt-get install ice icebox/trusty-zeroc icegrid/trusty-zeroc`

`sudo apt-get install libgoocanvasmm-2.0-dev`

And try to install all the packages/libraries which is showing not found while building if you still gets similar errors after it.

Kareem-Emad, 2019-03-31:
@anidh1997 if the installation above did not work for you -maybe something got messed up in linking *.so files-, you could try to put this file 
https://drive.google.com/open?id=1d90Cm2eWSr6fXBJ7-w5iKI0Ns43QLOiT
in
```
/usr/lib/x86_64-linux-gnu
```
and re-run cmake

anidh1997, 2019-04-01:
> @anidh1997 if the installation above did not work for you -maybe something got messed up in linking *.so files-, you could try to put this file
> https://drive.google.com/open?id=1d90Cm2eWSr6fXBJ7-w5iKI0Ns43QLOiT
> in
> 
> ```
> /usr/lib/x86_64-linux-gnu
> ```
> and re-run cmake

sleep-404, 2019-04-02:
Hi all, just finished editing a video on compiling JDE/base on a freshly installed Ubuntu 16.04 device.
[Compiling JDE/base from source (Updated)](https://www.youtube.com/watch?v=DslZX07qeeE)
Hope the above link helps someone.

closed 2019-05-23 by fqez

gonidelis, 2020-03-09:
@sleep-404  could you also add the commands that you follow? The website has changed.

sleep-404, 2020-03-09:
@Hi gonidelis , may I know what is the exact issue you are facing ? is it the same one that is posted above or something else, because this post is kind of old, so could you share your issue again please :)