JdeRobot/base #1351: JdeRobot Source Build issues

vinay0410, 2019-03-15:
Hi @fqez ,
I am just opening this to bring some issues to your attention.
First,
`sudo apt install jderobot-deps-dev` doesn't install ice dependencies and therefore `slice2py` isn't installed causing cmake error.
Although installing `jderobot-zeroc-ice-python` fixes this issue. Shouldn't this package be present in `jderobot-deps-dev`

Also, `src/drivers/gazebo/plugins/solo` doesn't contain any CMakeLists.txt, so either that directory can be initialized with an empty CMakeLists.txt or solo could be removed from plugins' `CMakeLists.txt`, because solo isn't being compiled now anyways

gonidelis, 2019-03-15:
I completely agree

fqez, 2019-03-15:
Hi @vinay0410,

thank you so much for summarizing the issues about this topic! About the solo driver, I fixed that this Thursday (creating an empty CMakeLists.txt as you suggest), so it shouldn't be a problem anymore. About the second one, I realized yesterday reviewing all the issues people are opening about this matter. 
I will try to fix this errors asap and release a new version of the project, so everyone can compile it without a problem. 

Sorry about the inconveniences guys, I will let you know when all this is solved.

sleep-404, 2019-03-17:
Hello @fqez  , 
Today I reinstalled Ubuntu 16.04 , and after installation when I'm trying to build I am getting the below error 
[image]
I have all the dependencies related to Opencv installed and also searched+installed using
`apt search opencv`
I searched for opencv in my computer using 
` pkg-config --libs opencv`
that returned many places , don't know which one to include for OpenCV_INCLUDE_DIR

sleep-404, 2019-03-17:
Sorry my bad, everything's working fine now :)

sleep-404, 2019-03-18:
After reinstalling ubuntu , I followed the exact same procedure as before , except I downloaded the repository that was given in the readme, and many errors popped up regarding missing dependencies even after installing jderobot-deps-dev, which was not the case earlier, when I cloned this particular repository instead the one mentioned in the readme though the readme directs me to this repository.

fqez, 2019-03-21:
Hi @sleep-404, I'm glad you got it working already :) Yeah, the docs are a bit out of date so we will try to update them as soon as possible to make the installation process as confortable as possible for you. I'm generating the proper issues to solve all this documentation problems.

Thank you so much for your feedback! It helps us to grow :)

gonidelis, 2019-03-21:
In the meantime how could I complete the installation ?

fqez, 2019-03-21:
Hi @gonidelis I highly recommend you to try the binary installation following the steps [here](https://jderobot.org/Installation#From_Debian_packages). If you are using Ubuntu 18.04 it should work without any problems

gonidelis, 2019-03-21:
Thank you a lot for your advise @fqez. The problem with binary installation is  that I cannot find the cameraserver path

closed 2021-02-21 by jmplaza