JdeRobot/base #1353: fatal error: interface/mmal/mmal.h: No such file or directory

gonidelis, 2019-03-20:
I followed the installation from source instructions and when I tried
`make` this log appeared:

> [ 65%] Building CXX object src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o
In file included from /home/john/base/src/drivers/picamserver/src/piCamCore.cpp:1:0:
/home/john/base/src/drivers/picamserver/src/piCamCore.h:3:33: **fatal error: interface/mmal/mmal.h: No such file or directory**
compilation terminated.
src/drivers/picamserver/CMakeFiles/picamserver.dir/build.make:110: recipe for target 'src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o' failed
make[2]: *** [src/drivers/picamserver/CMakeFiles/picamserver.dir/src/piCamCore.cpp.o] Error 1
CMakeFiles/Makefile2:5056: recipe for target 'src/drivers/picamserver/CMakeFiles/picamserver.dir/all' failed
make[1]: *** [src/drivers/picamserver/CMakeFiles/picamserver.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

Any idea how to solve it would be appreciated.
Thanks in advance

jmplaza, 2019-03-20:
picamsever should not be in the compilation chain of JdeRobot/base @fqez Picamserver is the "driver" for the camera in the RaspberryPi

gonidelis, 2019-03-21:
After removing **picamserver** installation was completed 100%.

closed 2019-03-21 by gonidelis